s60sc/ESP32-CAM_MJPEG2SD #257: No effect on clicking any buttons

CazYokoyama, 2023-08-12:
Since v8.7.3, common.js is introduced and MJPEG2SD.htm becomes smaller. When I click any buttons such as tools icon, nothing happens. Even more, no information is shown on bottom of the screen such as the attached screen shot even I wait for a long time such as 5 minutes.
[image]
Does anyone have the same symptom? or do I missing something?
v8.7.2 works fine.

s60sc, 2023-08-12:
Check app log

CazYokoyama, 2023-08-15:
How can I look at the app log? Clicking "Show Log" has nothing. Do you mean the message on the Serial port? Here it is. The line appears when I browse its IP address.

=============== ESP-CAM_MJPEG 8.7.4 ===============
[01:24:29.973 infoSD] SD card type SDHC, Size: 3789MB, Used space: 2MB, of total: 3780MB
[01:24:29.973 listFolder] Sketch size 1518kB
[01:24:30.265 listFolder] File: /data/configs.txt, size: 4016
[01:24:30.266 listFolder] SD_MMC: Total bytes 3964239872, Used bytes 2457600
[01:24:30.277 loadConfig] Load config
[01:24:30.353 updateAppStatus] Enabling motion detection
[01:24:30.355 printResetReason] Software reset via esp_restart
[01:24:30.356 printWakeupReason] Wakeup by reset
[01:24:30.563 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[01:24:30.844 setWifiSTA] Wifi Station IP from DHCP
.[01:24:30.948 onWiFiEvent] WiFi Station connection to TP-Link_A8B1, using hostname: ESP-CAM_MJPEG_A8DFD27EB994
..[01:24:31.929 onWiFiEvent] Wifi Station IP, use 'http://10.0.0.122' to connect
[01:24:32.475 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_A8DFD27EB994
[01:24:32.476 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_A8DFD27EB994 started, use 'http://192.168.4.1' to connect
[01:24:39.115 startWifi] Wifi stats for TP-Link_A8B1 - signal strength: -47 dBm; Encryption: WPA2_PSK; channel: 2
[01:24:39.116 startPing] Started ping monitoring
[01:24:39.129 startWebServer] Starting web server on port: 80
[01:24:39.131 startStreamServer] Starting streaming server on port: 82
[01:24:39.133 prepRecording] To record new AVI, do one of:
[01:24:39.143 prepRecording] - press Start Recording on web page
[01:24:39.144 prepRecording] - move in front of camera
[01:24:39.154 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[01:24:39.154 checkMemory] Free: heap 104800, block: 77812, pSRAM 2545779
[01:24:39.176 getLocalNTP] Using NTP server: pool.ntp.org
[01:24:39.177 showLocalTime] Got current time from NTP: 15/08/2023 01:24:39 with tz: GMT0
[01:24:39.206 wgetFile] Downloading /data/common.js from https://raw.githubusercontent.com/s60sc/ESP32-CAM_MJPEG2SD/master/data/common.js
[01:24:40.568 wgetFile] Downloaded /data/common.js, size 26697 bytes
[01:24:40.597 wgetFile] Downloading /data/MJPEG2SD.htm from https://raw.githubusercontent.com/s60sc/ESP32-CAM_MJPEG2SD/master/data/MJPEG2SD.htm
[01:24:43.059 wgetFile] Downloaded /data/MJPEG2SD.htm, size 79249 bytes
[01:24:43.077 resetWatchDog] WatchDog started using task: ping
[01:25:23.283 fileHandler] Sent /data/MJPEG2SD.htm to browser

s60sc, 2023-08-15:
looks ok, check browser inspector

CazYokoyama, 2023-08-15:
MJPEG2SD.htm  requests
c:/MyStuff/Arduino/CommonMaster/Latest/common.js:1  Failed to load resource: net::ERR_UNKNOWN_URL_SCHEME
/web?common.js:1  Failed to load resource: the server responded with a status of 400 (Bad Request)
as the attachment.
I am using Chrome on Ubuntu Linux.

[image]

s60sc, 2023-08-15:
the problem is:
/web?common.js:1 Failed to load resource: the server responded with a status of 400 (Bad Request)
which implies the file has not been stored - try using ota

CazYokoyama, 2023-08-15:
I copied data directory as
`~/ESP32-CAM_MJPEG2SD$ rsync -a data/ caz@diana2:/media/caz/ESP32-CAM_M/data/
`
Same.

When I browse http://10.0.0.122/web?common.js, I have "File does not exist or cannot be opened".

CazYokoyama, 2023-08-16:
The latest logs.

=============== ESP-CAM_MJPEG 8.7.4 ===============
[00:00:00.556 infoSD] SD card type SDHC, Size: 3789MB, Used space: 6MB, of total: 3780MB
[00:00:00.556 listFolder] Sketch size 1527kB
[00:00:00.852 listFolder] File: /data/configs.txt, size: 4028
[00:00:00.855 listFolder] File: /data/common.js, size: 26697
[00:00:00.858 listFolder] File: /data/MJPEG2SD.htm, size: 79249
[00:00:00.858 listFolder] SD_MMC: Total bytes 3964239872, Used bytes 7081984
[00:00:00.869 loadConfig] Load config
[00:00:00.947 updateAppStatus] Enabling motion detection
[00:00:00.950 printResetReason] Power on reset
[00:00:00.951 printWakeupReason] Wakeup by reset
[00:00:01.159 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:01.441 setWifiSTA] Wifi Station IP from DHCP
..[00:00:02.332 onWiFiEvent] WiFi Station disconnected
..[00:00:03.160 onWiFiEvent] WiFi Station connection to TP-Link_A8B1, using hostname: ESP-CAM_MJPEG_A8DFD27EB994
[00:00:03.193 onWiFiEvent] Wifi Station IP, use 'http://10.0.0.122' to connect
[00:00:03.491 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_A8DFD27EB994
[00:00:03.491 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_A8DFD27EB994 started, use 'http://192.168.4.1' to connect
[00:00:10.129 startWifi] Wifi stats for TP-Link_A8B1 - signal strength: -46 dBm; Encryption: WPA2_PSK; channel: 2
[00:00:10.130 startPing] Started ping monitoring
[00:00:10.144 getLocalNTP] Using NTP server: pool.ntp.org
[00:00:10.145 startWebServer] Starting web server on port: 80
[00:00:10.145 WARN getLocalNTP] Not yet synced with NTP 
[00:00:10.147 startStreamServer] Starting streaming server on port: 82
[00:00:10.158 prepRecording] To record new AVI, do one of:
[00:00:10.168 prepRecording] - press Start Recording on web page
[00:00:10.168 prepRecording] - move in front of camera
[00:00:10.179 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[00:00:10.179 checkMemory] Free: heap 68988, block: 42996, pSRAM 2545199
[00:00:10.187 resetWatchDog] WatchDog started using task: ping
[12:43:02.252 isNight] Night time
[12:43:19.995 fileHandler] Sent /data/MJPEG2SD.htm to browser
[12:43:30.268 getLocalNTP] Using NTP server: pool.ntp.org
[12:43:30.268 showLocalTime] Got current time from NTP: 16/08/2023 12:43:30 with tz: GMT0

Here is the message on console of developer tools.

DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/translations.js.map: System error: net::ERR_FILE_NOT_FOUND
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/cs-loader.js.map: System error: net::ERR_FILE_NOT_FOUND
(index):1970 
 GET c:/MyStuff/Arduino/CommonMaster/Latest/common.js net::ERR_UNKNOWN_URL_SCHEME
(index):1970 
 GET http://10.0.0.122/web?common.js net::ERR_ABORTED 400 (Bad Request)
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/common.js.map: System error: net::ERR_FILE_NOT_FOUND
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/translations.js.map: System error: net::ERR_FILE_NOT_FOUND
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/content-script-libs.js.map: System error: net::ERR_FILE_NOT_FOUND
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/frames.js.map: System error: net::ERR_FILE_NOT_FOUND
DevTools failed to load source map: Could not load content for chrome-extension://nbeldjopgciegccabfohnefghfpinncn/scripts/content-script.js.map: System error: net::ERR_FILE_NOT_FOUND
(index):1974 Uncaught ReferenceError: initialise is not defined
    at (index):1974:9
﻿
I.e. while the micro SD card has common.js in /data directory, http://10.0.0.122/web?common.js can't find it.

s60sc, 2023-08-16:
It could be a problem with your browser blocking js files.

CazYokoyama, 2023-08-16:
I confirm my Chrome has "Sites can use JavaScript" checked according to https://support.google.com/adsense/answer/12654?hl=en. I still can't have any action on clicking any button.
I give up. s60sc, thank you for your advices.

closed 2023-08-16 by CazYokoyama