s60sc/ESP32-CAM_MJPEG2SD #259: option to make files/folders readable

CazYokoyama, 2023-08-17:
I want to have a club website which

1. is browsed by club members
2. have ftp uploaded files

The ftp uploaded files are created as no read permission for group and others. Therefore the files are not browser accessible.
I created an patch, https://github.com/CazYokoyama/ESP32-CAM_MJPEG2SD/pull/2
However, making the files in readable for all is a security risk. Is there any way to give read permission only for my devices?

s60sc, 2023-08-17:
I can add a modified version of your patch in the next release as it only works for unix.

`Is there any way to give read permission only for my devices?`
You would need to create a suitable shell script on the ftp server.

CazYokoyama, 2023-08-18:
This is implemented by
local_umask=022
on vsftpd.conf. No need to modify ESP32-CAM_MJPEG2SD. Thank you anyway.

closed 2023-08-18 by CazYokoyama