s60sc/ESP32-CAM_MJPEG2SD #263: Unable to connect if # symbol is present in wifi password.

pig-cop, 2023-08-28:
My wifi password has a # symbol in it and it seems this causes the wifi to fail to connect.
If I change my wifi password in the router to not have a # symbol it works without a flaw.

s60sc, 2023-08-28:
'#' symbol has a special [meaning](https://en.wikipedia.org/wiki/URI_fragment) in URL so cant be used in a value sent from the web page

pig-cop, 2023-08-28:
It sure has a meaning, but it doesn't get url encoded when sending the value from the input form?
It's what is happening when one sends any text from any web page.

s60sc, 2023-08-29:
It would also do unwanted encoding.
You could modify the code to use post instead of get.

pig-cop, 2023-08-30:
I'll take a look, WPA/WPA2 PSK per standard is defined by a passphrase of 8 to 63 printable ASCII characters so I think is reasonable to be able to input a valid passphrase as in any network device.

closed 2023-08-30 by pig-cop

T1MOXA, 2025-07-31:
Same problem with # in AP name.