s60sc/ESP32-CAM_MJPEG2SD #264: [Enhancement] Is it possible to add FTPS?

lztz0022, 2023-08-29:
Sir s60sc, 
The project feature has been stable since 8.7.1a. I also see that the focus of your work is mainly shifted to the adaptation of ESP32S3.
Now I have a kind request for functional enhancement: many FTP service provider no longer provides pure FTP protocol, they think that plaintext transmission is too insecure, so they are now using FTP+TLS/SSL Explicit encryption. I learned that the ftplibpp library can be implemented, but it is too redundant. So, is it possible to implement TLS/SSL Explicit encryption enhancements based on the current FTP? Thank you and best regards.

s60sc, 2023-08-29:
it would require research which would not occur in the short term

s60sc, 2023-10-12:
tried, not successful

closed 2023-10-12 by s60sc

lztz0022, 2023-10-17:
It's regrettable, but thank you very much for trying.