s60sc/ESP32-CAM_MJPEG2SD #266: Pulling a still image via http?

aklstreak, 2023-08-31:
First of all, I am blown away with what you have done here! Unbelievable what you have crammed into that little board.

Question:  Is it possible to pull a still image via a http request? (like in the Arduino example code with http://ipaddress/capture)

Tried to search for that but no success so far :(

with best regards

s60sc, 2023-08-31:
Thanks

Click V icon on image view to download as jpeg to browser.

Or via URL from browser or player such as VLC:
-   single image: ipaddr:82/stream?random=1 (change number to avoid browser caching)
-   stream: ipaddr:82/stream?source=sensor 
-   select playback file: ipaddr/control?sfile=/20230830/20230830_124817_SVGA_20_6_120.avi 
  playback selected file: ipaddr:82/stream?source=file

aklstreak, 2023-08-31:
Perfect! Thank you so much for the quick response.  

I am pulling a jpg every couple of seconds into my home automation dashboard (iobroker). Works like a charm now.

closed 2023-08-31 by aklstreak

aklstreak, 2023-08-31:
Hi there,

 

would you mind if I post a message about your phantastic project in the iobroker community board? I am sure a number of people will be interested there.

And I don’t want to post anything without asking you first.

 

With best regards,

 

Albert

s60sc, 2023-08-31:
Fine by me 👍