s60sc/ESP32-CAM_MJPEG2SD #269: Por Forward

YaphethSN, 2023-09-09:
hello there! i would like to know how to use this code and port forward using my router only if that is possible, the code is working but im not sure how to port forward? if you could explain it or give a tutorial that would be magnificent, and im using a zte f609 router and not sure how to convert the lan to wan, sorry if im a bit slow

YaphethSN, 2023-09-09:
[]
[image]
(url)

heres what the port forward looks like in my router
doesnt match up with yours

s60sc, 2023-09-09:
https://www.cfos.de/en/cfos-personal-net/port-forwarding/zte-f609.htm
But dont use 80 for wan port

YaphethSN, 2023-09-10:
> https://www.cfos.de/en/cfos-personal-net/port-forwarding/zte-f609.htm But dont use 80 for wan port

what should i do then?

YaphethSN, 2023-09-10:
my main problem is the wan host star and end ip address, as i placed in the ip from whats my ip and it doesnt work
[image]

closed 2023-10-12 by s60sc