s60sc/ESP32-CAM_MJPEG2SD #271: continuous recording instead of motion or force start

johnpiker, 2023-09-14:
What changes should I made to start the recording once I switch on the device. As I don't want to the motion dependency and force start through start button.
I disable the motion sense but how to activate the continuous monitoring without the start on button. Can I hard code it ?

As I want continuous recording of 30 min simultaneously.
Please help..

s60sc, 2023-09-14:
in `mjpeg2sd.cpp`:
`forceRecord = true;` to start recording
`forceRecord = false;` to stop recording

johnpiker, 2023-09-15:
Thankyou for this,
Just one more doubt as where should i keep the record duration do that it will automatically start and stop recording simultaneously after every 30 min.
As I wanted to record the 30 min clips round the clock simultaneously.

s60sc, 2023-09-15:
`doAppPing() `in `appGlobals.cpp` is called every 30 secs, so put counter in that

johnpiker, 2023-09-15:
We have some customised requirement of  audio and video recordings using ESP32 boards and uploading the recorded files on server using internet. If you can provide us the complete solution for our requirement we can freelance this project on payment basis. Please share your Email ID .

johnpiker, 2023-09-15:
> doAppPing()

Thank you very much

s60sc, 2023-09-15:
> We have some customised requirement of audio and video recordings using ESP32 boards and uploading the recorded files on server using internet. If you can provide us the complete solution for our requirement we can freelance this project on payment basis. Please share your Email ID .

Thanks, but I only code as a hobby for my own interest.

johnpiker, 2023-09-18:
Sorry, and keep it on 👍

johnpiker, 2023-10-03:
static bool closeAvi() {
  // closes the recorded file
  uint32_t vidDuration = millis() - startTime;
  uint32_t vidDurationSecs = lround(vidDuration/1000.0);
  logLine();
  LOG_DBG("Capture time %u, min seconds: %u ", vidDurationSecs, minSeconds);

  cTime = millis();
  // write remaining frame content to SD
  aviFile.write(iSDbuffer, highPoint); 
  size_t readLen = 0;
  // add wav file if exists
  finishAudio(true);
  bool haveWav = haveWavFile();
  if (haveWav) {
    do {
      readLen = writeWavFile(iSDbuffer, RAMSIZE);
      aviFile.write(iSDbuffer, readLen);
    } while (readLen > 0);
  }
  // save avi index
  finalizeAviIndex(frameCnt);
  do {
    readLen = writeAviIndex(iSDbuffer, RAMSIZE);
    if (readLen) aviFile.write(iSDbuffer, readLen);
  } while (readLen > 0);
  // save avi header at start of file
  float actualFPS = (1000.0f * (float)frameCnt) / ((float)vidDuration);
  uint8_t actualFPSint = (uint8_t)(lround(actualFPS));  
  xSemaphoreTake(aviMutex, portMAX_DELAY);
  buildAviHdr(actualFPSint, fsizePtr, frameCnt);
  xSemaphoreGive(aviMutex); 
  aviFile.seek(0, SeekSet); // start of file
  aviFile.write(aviHeader, AVI_HEADER_LEN); 
  aviFile.close();
  LOG_DBG("Final SD storage time %lu ms", millis() - cTime);
  uint32_t hTime = millis(); 
  if (vidDurationSecs >= minSeconds) {
    // name file to include actual dateTime, FPS, duration, and frame count
    int alen = snprintf(aviFileName, FILE_NAME_LEN - 1, "%s_%s_%u_%u_%u%s.%s", 
      partName, frameData[fsizePtr].frameSizeStr, actualFPSint, vidDurationSecs, frameCnt, haveWav ? "_S" : "", FILE_EXT);
    if (alen > FILE_NAME_LEN - 1) LOG_WRN("file name truncated");
    SD_MMC.rename(AVITEMP, aviFileName);
    LOG_DBG("AVI close time %lu ms", millis() - hTime); 
    cTime = millis() - cTime;
    // AVI stats
    LOG_INF("******** AVI recording stats ********");
    LOG_ALT("Recorded %s", aviFileName);
    LOG_INF("AVI duration: %u secs", vidDurationSecs);
    LOG_INF("Number of frames: %u", frameCnt);
    LOG_INF("Required FPS: %u", FPS);
    LOG_INF("Actual FPS: %0.1f", actualFPS);
    LOG_INF("File size: %0.2f MB", (float)vidSize / ONEMEG);
    if (frameCnt) {
      LOG_INF("Average frame length: %u bytes", vidSize / frameCnt);
      LOG_INF("Average frame monitoring time: %u ms", dTimeTot / frameCnt);
      LOG_INF("Average frame buffering time: %u ms", fTimeTot / frameCnt);
      LOG_INF("Average frame storage time: %u ms", wTimeTot / frameCnt);
    }
    LOG_INF("Average SD write speed: %u kB/s", ((vidSize / wTimeTot) * 1000) / 1024);
    LOG_INF("File open / completion times: %u ms / %u ms", oTime, cTime);
    LOG_INF("Busy: %u%%", std::min(100 * (wTimeTot + fTimeTot + dTimeTot + oTime + cTime) / vidDuration, (uint32_t)100));
    checkMemory();
    LOG_INF("*************************************");
    if (mqtt_active) {
      sprintf(jsonBuff, "{\"RECORD\":\"OFF\", \"TIME\":\"%s\"}", esp_log_system_timestamp());
      mqttPublish(jsonBuff);
    }
    if (autoUpload) ftpFileOrFolder(aviFileName); // Upload it to remote ftp server if requested
    checkFreeSpace();
[... 71 lines cut ...]
    if (!isCapturing && wasCapturing) {
      // movement stopped
      finishRecording = true;
      if (lampAuto) setLamp(0); // switch off lamp
    }
    wasCapturing = isCapturing;
    LOG_DBG("============================");
  }
  if (fb != NULL) esp_camera_fb_return(fb);
  fb = NULL; 
  if (finishRecording) {
    // cleanly finish recording (normal or forced)
    
    if (stopPlayback) 
    Serial.println ("finish rec and stop playback");
    closeAvi();
    finishRecording = isCapturing = wasCapturing = stopPlayback = false; // allow for playbacks
  }
  return res;
}

I have done some changes in the process frame and close Avi in order to do the simlultaneous recording and i have given the timer as the number of frames as after certain numer of frames it should stop and save current recording and start new recording.

but the problem is that now i am not able to access the folder/file on the web page and not able to download them after connecting to ESP wifi.
can you help me with this

s60sc, 2023-10-03:
what do the logs say

johnpiker, 2023-10-05:
Nothing is showing on serial monitor while accessing the folder
[image]
[image]

closed 2023-10-12 by s60sc