s60sc/ESP32-CAM_MJPEG2SD #273: Can't connect via AP

GSternik, 2023-09-16:
I have that's logs from console:
[00:00:00.712 infoSD] SD card type SDHC, Size: 58.6GB
[00:00:00.712 listFolder] Sketch size 1.5MB
[00:00:01.005 listFolder] SD_MMC: 1024 bytes used
[00:00:01.006 loadConfig] Load config
[00:00:01.009 ERROR @ prefs.cpp:135] Failed to load file /data/configs.txt 
[00:00:01.022 WARN retrieveConfigVal] Key hostName not set 
[00:00:01.022 WARN retrieveConfigVal] Key AP_SSID not set 
[00:00:01.219 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:01.220 WARN retrieveConfigVal] Key framesize not set 
[00:00:01.486 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.501 WARN startWifi] SSID  not available, use AP 
[00:00:01.504 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_9993F92DE6B4
[00:00:01.505 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_9993F92DE6B4 started, use 'http://192.168.4.1' to connect
[00:00:04.529 startPing] Started ping monitoring - On
[00:00:04.534 startWebServer] Starting web server on port: 80
[00:00:04.536 startStreamServer] Starting streaming server on port: 82
[00:00:04.538 prepRecording] To record new AVI, do one of:
[00:00:04.548 prepRecording] - press Start Recording on web page
[00:00:04.548 prepRecording] - move in front of camera
 
[00:00:04.558 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[00:00:04.559 checkMemory] Free: heap 107612, block: 81908, pSRAM 2567431
[00:00:06.819 isNight] Night time
[00:00:09.530 resetWatchDog] WatchDog started using task: ping

I can not connect to 192.168.4.1. First versions of the ESP32-CAM_MJPEG2SD. My module is typic ESP-32-Cam with OV2640 camera.
I see access point in my computer network settings, when the AP is established connection to my module is not possible. I tried with different boards setting with library - ESP32 - v.2.0.09. - v.2.0.12

s60sc, 2023-09-16:
see #265

GSternik, 2023-09-16:
I notice this:
[00:00:01.009 ERROR @ prefs.cpp:135] Failed to load file /data/configs.txt

is this a problem?
All seems ok, I can connect with AP, but 192.168.4.1 is not available.
I tried char ST_SSID in utils.cpp and password. Module can not connect to router.
When I flashed with earlier version was ok. Now it can not work with older version and older esp32 board library.

GSternik, 2023-09-17:
I fixed this problem. Probably it was hardware settings of my computer. I tried to make connection with another computer. I can change AP connection to my router than it is working well. Thanks for help.
ps. I have the same board with this code and I can't start webserver, This is log:
ELF file SHA256: 240cbf561d4c9e26

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400de4dc: 0008e0c9 0000701d 5d004136
Core  1 register dump:
PC      : 0x400de4e0  PS      : 0x00060330  A0      : 0x8008ed8b  A1      : 0x3ffbcd30  
A2      : 0x3ffc4bdc  A3      : 0x00000008  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffbc878  A7      : 0x80000001  A8      : 0x800de4e0  A9      : 0x3ffbcd10  
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x80000000  
A14     : 0x007bf538  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Backtrace: 0x400de4dd:0x3ffbcd30 0x4008ed88:0x3ffbcd50

Where is the problem?

GSternik, 2023-09-17:
After change flash frequency to 40MHz  camera continuously rebooting :
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13936
load:0x40080400,len:3600
entry 0x400805f0
E (576) esp_core_dump_flash: 枮"յ⸮⸮data check failed:
Calculated checksum='964f8b52'
Image checksum='3ad11fdf'
=============== ESP-CAM_MJPEG 8.8 ===============
[00:00:08.306 infoSD] SD card type SDHC, Size: 29.7GB
E (819) esp_image: Checksum failed. Calculated 0xa read 0x33
[00:00:08.307 listFolder] Sketch size 0 bytes
[00:00:08.607 listFolder] File: /data/configs.txt, size: 4191 bytes
[00:00:08.612 listFolder] File: /data/common.js, size: 26704 bytes
[00:00:08.616 listFolder] File: /data/MJPEG2SD.htm, size: 78631 bytes
[00:00:08.616 listFolder] SD_MMC: 28.8GB used
[00:00:08.627 loadConfig] Load config
[00:00:08.703 updateAppStatus] Disabling motion detection
[00:00:08.706 printResetReason] Software reset due to exception/panic
[00:00:08.706 printWakeupReason] Wakeup by reset
E (971) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
[00:00:09.148 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
abort() was called at PC 0x40091aeb on core 0
Backtrace: 0x40084319:0x3ffde8d0 0x40091af5:0x3ffde8f0 0x400981a9:0x3ffde910 0x40091aeb:0x3ffde990 0x40145614:0x3ffde9b0 0x401456c1:0x3ffde9e0 0x401790ea:0x3ffdea00 0x401798cb:0x3ffdea20 0x40177ee4:0x3ffdea40 0x401dcd1d:0x3ffdea60
ELF file SHA256: 56b0bbab9be50ddb

J-SIT, 2023-09-27:
Hello all.
I also had problems that it wanted to connect to my AP again and again. I found out that it was probably due to the special characters in my WLAN password. But with letters, numbers and minus it works fine.

Many greetings

Julian

closed 2023-10-12 by s60sc