s60sc/ESP32-CAM_MJPEG2SD #276: FTP feature uploads slowly

lztz0022, 2023-09-23:
Sir, I am using version 8.7.1 now, and I find that the FTP upload speed has become very slow, as follows: 
```
[11:48:20.398 ftpStoreFile] Upload file: 20230921_105509_XGA_3_2_120_T.avi, size: 5.1MB
[11:48:25.822 ftpStoreFile] Uploaded 3%
[11:48:30.964 ftpStoreFile] Uploaded 7%
[11:48:36.104 ftpStoreFile] Uploaded 11%
[11:48:42.916 ftpStoreFile] Uploaded 15%
[11:48:48.140 ftpStoreFile] Uploaded 19%
[11:48:53.248 ftpStoreFile] Uploaded 22%
[11:48:58.405 ftpStoreFile] Uploaded 26%
[11:49:03.452 ftpStoreFile] Uploaded 30%
[11:49:10.053 ftpStoreFile] Uploaded 34%
[11:49:15.173 ftpStoreFile] Uploaded 38%
[11:49:20.295 ftpStoreFile] Uploaded 42%
[11:49:25.434 ftpStoreFile] Uploaded 45%
[11:49:30.555 ftpStoreFile] Uploaded 49%
[11:49:37.191 ftpStoreFile] Uploaded 53%
[11:49:42.311 ftpStoreFile] Uploaded 57%
[11:49:47.512 ftpStoreFile] Uploaded 61%
[11:49:52.568 ftpStoreFile] Uploaded 65%
[11:49:57.707 ftpStoreFile] Uploaded 68%
[11:50:03.408 ftpStoreFile] Uploaded 72%
[11:50:09.347 ftpStoreFile] Uploaded 76%
[11:50:14.465 ftpStoreFile] Uploaded 80%
[11:50:19.497 ftpStoreFile] Uploaded 84%
[11:50:24.616 ftpStoreFile] Uploaded 87%
[11:50:29.828 ftpStoreFile] Uploaded 91%
[11:50:36.790 ftpStoreFile] Uploaded 95%
[11:50:42.013 ftpStoreFile] Uploaded 99%
[11:50:43.037 ftpStoreFile] Uploaded 5.1MB in 142 sec
```
It took 142 seconds to upload a 5.1MB file. The same FTP server and files can be uploaded with WinSCP in less than 10 seconds. I remember the previous versions of FTP upload speed was fast.

s60sc, 2023-09-23:
My cam does it in 8 secs, you have a slow connection:
```
[09:02:01.972 ftpStoreFile] Upload file: 20230923_085313_SVGA_20_8_160.avi, size: 5.0MB
[09:02:02.514 ftpStoreFile] Uploaded 5%
[09:02:02.933 ftpStoreFile] Uploaded 10%
[09:02:03.366 ftpStoreFile] Uploaded 15%
[09:02:03.817 ftpStoreFile] Uploaded 20%
[09:02:04.301 ftpStoreFile] Uploaded 25%
[09:02:04.756 ftpStoreFile] Uploaded 30%
[09:02:05.177 ftpStoreFile] Uploaded 35%
[09:02:05.586 ftpStoreFile] Uploaded 40%
[09:02:06.008 ftpStoreFile] Uploaded 45%
[09:02:06.472 ftpStoreFile] Uploaded 50%
[09:02:06.938 ftpStoreFile] Uploaded 55%
[09:02:07.358 ftpStoreFile] Uploaded 60%
[09:02:07.787 ftpStoreFile] Uploaded 65%
[09:02:08.266 ftpStoreFile] Uploaded 70%
[09:02:08.691 ftpStoreFile] Uploaded 75%
[09:02:09.120 ftpStoreFile] Uploaded 80%
[09:02:09.560 ftpStoreFile] Uploaded 85%
[09:02:10.011 ftpStoreFile] Uploaded 90%
[09:02:10.455 ftpStoreFile] Uploaded 95%
[09:02:10.895 ftpStoreFile] Uploaded 100%
[09:02:10.989 ftpStoreFile] Uploaded 5.0MB in 8 sec
```

lztz0022, 2023-09-23:
Okay, I'll check the device and my FTP server again.
But a few months ago, the upload speed was still very fast, maybe it was a problem with my network.
Another thing, the error I encountered in [#210](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/210) recurred today:
```
[00:00:08.205 startPing] Started ping monitoring
[00:00:08.210 startWebServer] Starting web server on port: 80
[00:00:08.214 getLocalNTP] Using NTP server: pool.ntp.org
[08:00:08.215 WARN getLocalNTP] Not yet synced with NTP 
[08:00:08.215 startStreamServer] Starting streaming server on port: 81
[08:00:08.275 prepRecording] To record new AVI, do one of:
Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x40091d26  PS      : 0x00040b36  A0      : 0x80086f38  A1      : 0x3ffe53a0  
A2      : 0x3ffb02ec  A3      : 0x00060b23  A4      : 0x3ffb8000  A5      : 0x3f42fb50  
A6      : 0x00000003  A7      : 0xff000000  A8      : 0x40080080  A9      : 0x3ffe5490  
A10     : 0x00060d36  A11     : 0x00040026  A12     : 0x00000004  A13     : 0x00000000  
A14     : 0x3ffb6884  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000002  
EXCVADDR: 0xffffffe0  LBEG    : 0x4008dedd  LEND    : 0x4008deed  LCOUNT  : 0xfffffffa  

Backtrace: 0x40091d23:0x3ffe53a0 0x40086f35:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 0x40091d23:0x3ffe5450 0x40091d23:0x3ffe5470 0x40091d23:0x3ffe5490 0x4008007d:0x3ffe53a0 0x4008432f:0x3ffe53c0 0x40091d23:0x3ffe53f0 0x40091d23:0x3ffe5410 0x40091d23:0x3ffe5430 |<-CONTINUES

ELF file SHA256: 726e65506c3a92c1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (533) esp_core_dump_f�f��� Core dump data check failed:
Calculated checksum='b6522da6'
Image checksum='ffffffff'

[00:00:09.668 print_wakeup_reason] Wakeup by reset
[00:00:09.668 setup] =============== Starting ===============
```

closed 2023-10-12 by s60sc