s60sc/ESP32-CAM_MJPEG2SD #277: Restart scheduler

DiBochev, 2023-09-23:
Hi all,
This software is great. The only issue is the crappy hardware, because it overheats and needs ALOT of heatsinks.
Can you create a restart scheduler policy: for example I want every day at 1:00 AM to restart the device?

Many thanks ion advantage :)

s60sc, 2023-09-23:
add following as first line in function `doAppPing() `in file `appSpecific.cpp`:

`  if (checkAlarm(1)) doRestart("User alarm"); // where 1 is 1am`

lztz0022, 2023-09-23:
> add following as first line in function `doAppPing() `in file `appSpecific.cpp`:
> 
> ` if (checkAlarm(1)) doRestart("User alarm"); // where 1 is 1am`

What if it's 11:30 PM?

DiBochev, 2023-09-25:
Thank you very much for the fast answer, but can this be made configurable using the web page?
For example restart every X hours (1, 2,3,.... 12) or restart at current time daily, weekly, monthly?

closed 2023-10-12 by s60sc