s60sc/ESP32-CAM_MJPEG2SD #280: Stored parameters are not used

DeepooBelief, 2023-10-09:
Hi, 
I have a problem with your application. When I change some settings in the web page, like setting hmirror to 1 and expected fps to 30. when I power off and turn it on again, everything in the web page is correct, hmirror is 1 and fps is 30. but in the data stream, the camera is not mirrored and the expected fps is 20, which is the same as default settings. In addition, the time zone parameter does not appear to be stored or used after a reboot.
Is there something I haven't configured correctly?
Thank you.

s60sc, 2023-10-09:
Press Save Settings to make changes permanent and refresh web page

DeepooBelief, 2023-10-09:
> Press Save Settings to make changes permanent and refresh web page

It seems that the problem is not solved
Steps to reproduce:
1. Open the web page
2. change fps to 15
3. press "save settings"
4. restart esp32

And log from serial monitor (the default fps for SVGA was changed to 25 in appGlobals.h):
22:28:07.668 -> [22:28:07.512 fileHandler] Sent /data/MJPEG2SD.htm to browser
22:28:17.661 -> [22:28:17.469 savePrefs] Saved preferences
22:28:17.661 -> [22:28:17.489 saveConfigVect] Config file saved 
22:28:20.560 -> [22:28:20.374 processFrame] Capture started by Motion  
22:28:20.701 -> .................................................. 
22:28:23.811 -> .................................................. 
22:28:27.161 -> .................... 
22:28:28.603 -> [22:28:28.426 closeAvi] ******** AVI recording stats ********
22:28:28.603 -> [22:28:28.426 closeAvi] Recorded /20231009/20231009_222820_SVGA_15_8_120.avi 
22:28:28.603 -> [22:28:28.436 closeAvi] AVI duration: 8 secs
22:28:28.603 -> [22:28:28.437 closeAvi] Number of frames: 120
22:28:28.603 -> [22:28:28.437 closeAvi] Required FPS: 15
22:28:28.603 -> [22:28:28.447 closeAvi] Actual FPS: 15.0
22:28:28.603 -> [22:28:28.448 closeAvi] File size: 2.2MB
22:28:28.650 -> [22:28:28.448 closeAvi] Average frame length: 19057 bytes
22:28:28.650 -> [22:28:28.458 closeAvi] Average frame monitoring time: 2 ms
22:28:28.650 -> [22:28:28.458 closeAvi] Average frame buffering time: 0 ms
22:28:28.650 -> [22:28:28.469 closeAvi] Average frame storage time: 10 ms
22:28:28.650 -> [22:28:28.469 closeAvi] Average SD write speed: 1744 kB/s
22:28:28.650 -> [22:28:28.479 closeAvi] File open / completion times: 3 ms / 57 ms
22:28:28.650 -> [22:28:28.479 closeAvi] Busy: 20%
22:28:28.650 -> [22:28:28.490 checkMemory] Free: heap 111008, block: 77812, pSRAM 2126811
22:28:28.650 -> [22:28:28.490 closeAvi] *************************************
22:28:28.698 -> [22:28:28.501 checkFreeSpace] Storage free space: 13140MB
22:28:37.806 -> ets Jul 29 2019 12:21:46
22:28:37.806 -> 
22:28:37.806 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:28:37.806 -> configsip: 0, SPIWP:0xee
22:28:37.806 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:28:37.806 -> mode:DIO, clock div:1
22:28:37.853 -> load:0x3fff0030,len:1344
22:28:37.853 -> load:0x40078000,len:13964
22:28:37.853 -> load:0x40080400,len:3600
22:28:37.853 -> entry 0x400805f0
22:28:38.825 -> [   481][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
22:28:38.825 -> 
22:28:38.825 -> 
22:28:38.825 -> =============== ESP-CAM_MJPEG 8.8 ===============
22:28:38.872 -> [00:00:00.556 infoSD] SD card type SDHC, Size: 14.4GB
22:28:39.154 -> [00:00:00.557 listFolder] Sketch size 1.5MB
22:28:39.200 -> [00:00:00.858 listFolder] File: /data/configs.txt, size: 4207 bytes
22:28:39.200 -> [00:00:00.861 listFolder] File: /data/common.js, size: 26704 bytes
22:28:39.200 -> [00:00:00.863 listFolder] File: /data/MJPEG2SD.htm, size: 78631 bytes
22:28:39.200 -> [00:00:00.874 listFolder] SD_MMC: 1.6GB used
22:28:39.200 -> [00:00:00.875 loadConfig] Load config
22:28:39.293 -> [00:00:00.954 updateAppStatus] Enabling motion detection
22:28:39.293 -> [00:00:00.956 setupLamp] Setup Lamp Led for ESP32 Cam board
22:28:39.293 -> [00:00:00.958 printResetReason] Power on reset
22:28:39.293 -> [00:00:00.959 printWakeupReason] Wakeup by reset
22:28:39.478 -> [00:00:01.169 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
22:28:39.754 -> [00:00:01.440 setWifiSTA] Wifi Station IP from DHCP
22:28:39.800 -> ....[00:00:03.189 onWiFiEvent] WiFi Station disconnected
22:28:41.798 -> .[00:00:03.901 onWiFiEvent] WiFi Station connection to test, using hostname: ESP-CAM_MJPEG_5CBA98F9D108
22:28:42.263 -> [00:00:03.963 onWiFiEvent] Wifi Station IP, use 'http://192.168.43.128' to connect
22:28:42.310 -> [00:00:03.982 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_5CBA98F9D108
22:28:42.310 -> [00:00:03.982 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_5CBA98F9D108 started, use 'http://192.168.4.1' to connect
22:28:49.164 -> [00:00:10.821 startWifi] Wifi stats for test - signal strength: -48 dBm; Encryption: WPA2_PSK; channel: 1
22:28:49.164 -> [00:00:10.829 startPing] Started ping monitoring - On
22:28:49.164 -> [00:00:10.833 startWebServer] Starting web server on port: 80
22:28:49.164 -> [00:00:10.835 startStreamServer] Starting streaming server on port: 82
[... 11 lines cut ...]
22:29:04.315 -> [22:29:03.963 fileHandler] Sent /data/MJPEG2SD.htm to browser
22:29:09.236 -> [22:29:08.859 processFrame] Capture started by Motion  
22:29:09.236 -> .................................................. 
22:29:11.230 -> .................................................. 
22:29:13.234 -> .................................................. 
22:29:15.231 ->  
22:29:15.277 -> [22:29:14.905 closeAvi] ******** AVI recording stats ********
22:29:15.277 -> [22:29:14.906 closeAvi] Recorded /20231009/20231009_222908_SVGA_25_6_150.avi 
22:29:15.277 -> [22:29:14.916 closeAvi] AVI duration: 6 secs
22:29:15.277 -> [22:29:14.916 closeAvi] Number of frames: 150
22:29:15.277 -> [22:29:14.917 closeAvi] Required FPS: 25
22:29:15.277 -> [22:29:14.927 closeAvi] Actual FPS: 25.0
22:29:15.324 -> [22:29:14.927 closeAvi] File size: 2.7MB
22:29:15.324 -> [22:29:14.928 closeAvi] Average frame length: 18613 bytes
22:29:15.324 -> [22:29:14.938 closeAvi] Average frame monitoring time: 1 ms
22:29:15.324 -> [22:29:14.938 closeAvi] Average frame buffering time: 0 ms
[... cut ...]  frame storage time: 9 ms
22:29:15.324 -> [22:29:14.949 closeAvi] Average SD write speed: 1888 kB/s
22:29:15.324 -> [22:29:14.959 closeAvi] File open / completion times: 3 ms / 44 ms
22:29:15.324 -> [22:29:14.959 closeAvi] Busy: 28%
22:29:15.324 -> [22:29:14.959 checkMemory] Free: heap 111328, block: 73716, pSRAM 2126419
22:29:15.324 -> [22:29:14.970 closeAvi] *************************************
22:29:15.372 -> [22:29:14.981 checkFreeSpace] Storage free space: 13137MB

Thank you for your help.

s60sc, 2023-10-09:
Dont change [appGlobals.h](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/blob/74d0d0b8e2733dc59151681532a12d7a47dd6d98/appGlobals.h#L47-L48)
Use the web interface.

DeepooBelief, 2023-10-09:
Hello.
Now I used the origin code, but problem still exist:

23:16:34.294 -> [23:16:34.259 savePrefs] Saved preferences
23:16:34.328 -> [23:16:34.278 saveConfigVect] Config file saved 
23:16:38.590 -> [23:16:38.541 processFrame] Capture started by Motion  
23:16:38.760 -> .............................. 
23:16:39.882 -> .................................................. 
23:16:41.872 -> ............................ 
23:16:43.069 -> [23:16:43.004 closeAvi] Insufficient capture duration: 4 secs
23:16:45.431 -> [23:16:45.378 processFrame] Capture started by Motion  
23:16:45.431 -> ...................... 
23:16:46.253 -> .................................................. 
23:16:48.268 -> .................................................. 
23:16:50.249 -> .................................................. 
23:16:52.267 -> ............................................ 
23:16:54.111 -> [23:16:54.062 closeAvi] ******** AVI recording stats ********
23:16:54.111 -> [23:16:54.062 closeAvi] Recorded /20231009/20231009_231645_SVGA_25_9_216.avi 
23:16:54.111 -> [23:16:54.072 closeAvi] AVI duration: 9 secs
23:16:54.111 -> [23:16:54.073 closeAvi] Number of frames: 216
23:16:54.146 -> [23:16:54.073 closeAvi] Required FPS: 27
23:16:54.146 -> [23:16:54.083 closeAvi] Actual FPS: 25.0
23:16:54.146 -> [23:16:54.084 closeAvi] File size: 4.0MB
23:16:54.146 -> [23:16:54.084 closeAvi] Average frame length: 19236 bytes
23:16:54.146 -> [23:16:54.095 closeAvi] Average frame monitoring time: 30 ms
23:16:54.146 -> [23:16:54.095 closeAvi] Average frame buffering time: 0 ms
23:16:54.146 -> [23:16:54.105 closeAvi] Average frame storage time: 9 ms
23:16:54.146 -> [23:16:54.105 closeAvi] Average SD write speed: 2072 kB/s
23:16:54.180 -> [23:16:54.116 closeAvi] File open / completion times: 3 ms / 46 ms
23:16:54.180 -> [23:16:54.116 closeAvi] Busy: 100%
23:16:54.180 -> [23:16:54.116 checkMemory] Free: heap 111884, block: 77812, pSRAM 2126031
23:16:54.180 -> [23:16:54.127 closeAvi] *************************************
23:16:54.180 -> [23:16:54.137 checkFreeSpace] Storage free space: 13107MB
23:17:03.118 -> [23:17:03.053 processFrame] Capture started by Motion  
23:17:03.118 -> ...... 
23:17:03.290 -> .............................ets Jul 29 2019 12:21:46
23:17:04.615 -> 
23:17:04.615 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
23:17:04.615 -> configsip: 0, SPIWP:0xee
23:17:04.615 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:17:04.649 -> mode:DIO, clock div:1
23:17:04.649 -> load:0x3fff0030,len:1344
23:17:04.649 -> load:0x40078000,len:13964
23:17:04.649 -> load:0x40080400,len:3600
23:17:04.649 -> entry 0x400805f0
23:17:05.606 -> [   481][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
23:17:05.640 -> 
23:17:05.640 -> 
23:17:05.640 -> =============== ESP-CAM_MJPEG 8.8 ===============
23:17:05.674 -> [00:00:00.557 infoSD] SD card type SDHC, Size: 14.4GB
23:17:05.980 -> [00:00:00.558 listFolder] Sketch size 1.5MB
23:17:05.980 -> [00:00:00.859 listFolder] File: /data/configs.txt, size: 4207 bytes
23:17:05.980 -> [00:00:00.862 listFolder] File: /data/common.js, size: 26704 bytes
23:17:06.014 -> [00:00:00.864 listFolder] File: /data/MJPEG2SD.htm, size: 78631 bytes
23:17:06.014 -> [00:00:00.875 listFolder] SD_MMC: 1.6GB used
23:17:06.014 -> [00:00:00.875 loadConfig] Load config
23:17:06.083 -> [00:00:00.955 updateAppStatus] Enabling motion detection
23:17:06.083 -> [00:00:00.956 setupLamp] Setup Lamp Led for ESP32 Cam board
23:17:06.083 -> [00:00:00.958 printResetReason] Power on reset
23:17:06.083 -> [00:00:00.958 printWakeupReason] Wakeup by reset
23:17:06.289 -> [00:00:01.166 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
23:17:06.562 -> [00:00:01.431 setWifiSTA] Wifi Station IP from DHCP
23:17:06.596 -> ..[00:00:02.167 onWiFiEvent] WiFi Station disconnected
23:17:07.588 -> .[00:00:02.894 onWiFiEvent] WiFi Station connection to test, using hostname: ESP-CAM_MJPEG_5CBA98F9D108
23:17:08.066 -> [00:00:02.927 onWiFiEvent] Wifi Station IP, use 'http://192.168.43.128' to connect
23:17:08.101 -> [00:00:02.971 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_5CBA98F9D108
23:17:08.101 -> [00:00:02.971 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_5CBA98F9D108 started, use 'http://192.168.4.1' to connect
23:17:14.942 -> [00:00:09.810 startWifi] Wifi stats for test - signal strength: -27 dBm; Encryption: WPA2_PSK; channel: 1
23:17:14.942 -> [00:00:09.811 startPing] Started ping monitoring - On
23:17:14.942 -> [00:00:09.825 startWebServer] Starting web server on port: 80
[... 8 lines cut ...]
23:17:15.252 -> [00:00:10.123 resetWatchDog] WatchDog started using task: ping
23:17:15.252 -> [00:00:10.123 getLocalNTP] Using NTP server: pool.ntp.org
23:17:15.252 -> [00:00:10.124 WARN getLocalNTP] Not yet synced with NTP 
23:17:15.490 -> [23:35:08.646 showLocalTime] Got current time from browser: 09/10/2023 23:35:08 with tz: GMT0BST,M3.5.0/1,M10.5.0
23:17:18.638 -> [23:35:11.785 processFrame] Capture started by Motion  
23:17:18.638 -> .................................................. 
23:17:21.064 -> .................................................. 
23:17:23.591 -> .................... 
23:17:24.648 -> [23:17:24.631 closeAvi] ******** AVI recording stats ********
23:17:24.681 -> [23:17:24.631 closeAvi] Recorded /20231009/20231009_233511_SVGA_20_6_120.avi 
23:17:24.681 -> [23:17:24.642 closeAvi] AVI duration: 6 secs
23:17:24.681 -> [23:17:24.642 closeAvi] Number of frames: 120
[... cut ...]
23:17:24.715 -> [23:17:24.674 closeAvi] Average frame storage time: 7 ms
23:17:24.715 -> [23:17:24.674 closeAvi] Average SD write speed: 1964 kB/s
23:17:24.715 -> [23:17:24.685 closeAvi] File open / completion times: 3 ms / 43 ms
23:17:24.715 -> [23:17:24.685 closeAvi] Busy: 17%
23:17:24.715 -> [23:17:24.685 checkMemory] Free: heap 111408, block: 77812, pSRAM 2126811
23:17:24.749 -> [23:17:24.696 closeAvi] *************************************
23:17:24.749 -> [23:17:24.706 checkFreeSpace] Storage free space: 13106MB

s60sc, 2023-10-09:
After save, on browser enter `your_ip/web?configs.txt` and check the fps entry

DeepooBelief, 2023-10-09:
config.txt is corrent:
I think config.txt is correct, it gives "fps~27~98~~na" when I set fps to 27.
What part of the code reads the camera configuration parameters such as fps from config.txt or perferences? Maybe there's something wrong with that part.

s60sc, 2023-10-09:
Hmm, there could be a bug, I'll check tomorrow.

s60sc, 2023-10-10:
Sorry, was a bug, should be fixed.

s60sc, 2023-10-11:
I'll add a fix to the next update

s60sc, 2023-10-12:
In v9.0

closed 2023-10-12 by s60sc