s60sc/ESP32-CAM_MJPEG2SD #289: MJPEG2SD.htm not working for stable version

anhydrousmule, 2023-10-13:
Auto-download fetches too new/updated version of .htm to sdcard and that references to "/sustain?" uri's, that functionality only exists in latest/unstable version?. Result is that web control buttons for streaming, playback etc do not work.
Also streamserver port definition in .htm is changed to 81, which is not working it needs to be 82 ?

Dirty quickfix to make stream button working again;

In function "appInit()" modify "streamServer" variable initialization to go one port number higher;
// streamServer= webHost + ':' + String(+webPort + 1);
streamServer= webHost + ':' + String(+webPort + 2);

In function "activateStreamButton()" modify "view.src" initialization to use old way of functionality;
// view.src = streamServer + '/sustain?stream=1';
view.src = streamServer + '/stream?source=sensor';

I managed to get all other buttons working by modifying their uri's too, but for download button my limited debugging skills or lack of sleep failed me.

First time reporting issue so thank you for your marvellous code and understanding.

s60sc, 2023-10-14:
unforseen problem that v8.8 will auto download v9.0 data folder so manually instal v8.8 data folder.

anhydrousmule, 2023-10-18:
Where to get the "right" 8.8 folder? At least the files that are included with 8.8 code download are not working either... they seem to be too old version for 8.8 and web control is broken even more with them?

s60sc, 2023-10-18:
Copy the entire 8.8 data folder to the sd card

closed 2023-11-18 by s60sc