s60sc/ESP32-CAM_MJPEG2SD #292: Cannot open email attachment

wdouglas1999, 2023-10-17:
Motion is detected and email is send through gmail smtp. SD card is recording fine. Email is received but the attachment will not open. I am using Freenove ESP 32 WROOM board.
V9.0 of the code.
Thank you for taking the time to create this project. It is quite awesome.

s60sc, 2023-10-18:
Thanks. I cannot diagnose problems outside the app

wdouglas1999, 2023-10-18:
Thank you for the reply. I was just wondering if maybe the attached image was being corrupted.
I will gladly diagnose further. Could I ask you please to verify that you receive the attachments and are able to open them? Thank you again.

wdouglas1999, 2023-10-18:
Thanks, I will have a look further. Much appreciated.

closed 2023-10-18 by wdouglas1999