s60sc/ESP32-CAM_MJPEG2SD #293: Telegram integration possible?

dr-garimella, 2023-10-18:
Your work is really excellent. Really, I don't have any words to express the gratitude. 

Small doubt if possible.! 

Is it possible to access (open) the SD card files through telegram group.?! 

Regards.

s60sc, 2023-10-18:
It is something I will investigate in the future

FBMinis, 2023-10-25:
> Your work is really excellent. Really, I don't have any words to express the gratitude.
> 
> Small doubt if possible:
> 
> Is it possible to access (open) the SD card files through telegram group.?!
> 
> Regards.

In the meantime, I would suggest you try this application which send a photo and video to telegram when movement is detected by a PIR connected to GPIO13: https://github.com/jameszah/ESP32-CAM-Video-Telegram

dr-garimella, 2023-10-25:
> In the meantime, I would suggest you try this application which send a photo and video to telegram when movement is detected by a PIR connected to GPIO13: [...]

Thank you for the response. 

However, it doesn't work. Already tried the code. Videos are not sending properly. The code is good to send Photos alone.

FBMinis, 2023-10-25:
> However, it doesn't work. Already tried the code. Videos are not sending properly. The code is good to send Photos alone.

I have had 3 cameras working outside for over 12 months without a problem, they send video and photo when movement is detected.

dr-garimella, 2023-10-25:
Ohh.! May be my hard luck. I tried many times and finally, left it.

dr-garimella, 2023-10-25:
> > However, it doesn't work. Already tried the code. Videos are not sending properly. The code is good to send Photos alone.
> 
> I have had 3 cameras working outside for over 12 months without a problem, they send video and photo when movement is detected.

Ohh.! May be my hard luck. I tried many times and finally, left it.

FBMinis, 2023-10-25:
I put this together, it sends a photo to Telegram when haveMotion() returns true, one photo per 10sec interval at most, so not to flood the connection.

```
#include <WiFi.h>
#include <AsyncTelegram2.h>
#include "esp_camera.h"
#include "soc/soc.h"           // Brownout error fix
#include "soc/rtc_cntl_reg.h"  // Brownout error fix

#include <WiFiClientSecure.h>
WiFiClientSecure secure_client;

const char* ssid = "***********";                                        // SSID WiFi network
const char* pass = "***********";                                        // Password  WiFi network
const char* token = "***********";  // Telegram token
int64_t userid = ***********;

// Timezone definition to get properly time from NTP server
#define MYTZ "WET0WEST,M3.5.0/1,M10.5.0"

AsyncTelegram2 myBot(secure_client);

#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27

#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
#define Y3_GPIO_NUM 18
#define Y2_GPIO_NUM 5
#define VSYNC_GPIO_NUM 25
#define HREF_GPIO_NUM 23
#define PCLK_GPIO_NUM 22

static camera_config_t camera_config = {
  .pin_pwdn = PWDN_GPIO_NUM,
  .pin_reset = RESET_GPIO_NUM,
  .pin_xclk = XCLK_GPIO_NUM,
  .pin_sscb_sda = SIOD_GPIO_NUM,
  .pin_sscb_scl = SIOC_GPIO_NUM,
  .pin_d7 = Y9_GPIO_NUM,
  .pin_d6 = Y8_GPIO_NUM,
  .pin_d5 = Y7_GPIO_NUM,
  .pin_d4 = Y6_GPIO_NUM,
  .pin_d3 = Y5_GPIO_NUM,
  .pin_d2 = Y4_GPIO_NUM,
  .pin_d1 = Y3_GPIO_NUM,
  .pin_d0 = Y2_GPIO_NUM,
  .pin_vsync = VSYNC_GPIO_NUM,
  .pin_href = HREF_GPIO_NUM,
  .pin_pclk = PCLK_GPIO_NUM,
  .xclk_freq_hz = 20000000,
  .ledc_timer = LEDC_TIMER_0,
  .ledc_channel = LEDC_CHANNEL_0,
  .pixel_format = PIXFORMAT_JPEG,  /* PIXFORMAT_RGB565, // for face detection/recognition */
  .frame_size = FRAMESIZE_VGA,
  .jpeg_quality = 12,
  .fb_count = 1,
  .fb_location = CAMERA_FB_IN_PSRAM,
  .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
#if CONFIG_CAMERA_CONVERTER_ENABLED
  .conv_mode = CONV_DISABLE,  /*!< RGB<->YUV Conversion mode */
#endif
  //.sccb_i2c_port = 1          /*!< If pin_sccb_sda is -1, use the already configured I2C bus by number */
[... 207 lines cut ...]
    haveMotion = checkMotion(fb, haveMotion);

    if (haveMotion) {
      haveMotion = false;
      
      Serial.println("movement detected");

      unsigned long currentTime = millis();

      if (currentTime - lastPictureTime >= interval) {
        lastPictureTime = currentTime;
        Serial.println("\nSending Photo from CAM");
        if (sendPicture(userid))
          Serial.println("Picture sent successfull");
        else
          myBot.sendTo(userid, "Error, picture not sent.");
      }
    }
    
    esp_camera_fb_return(fb);
  } else Serial.println("Failed to get frame");
  delay(100);
}

```

dr-garimella, 2023-10-26:
> I put this together, it sends a photo to Telegram when haveMotion() returns true, one photo per 10sec interval at most, so not to flood the connection.
>
> ```
> #include <WiFi.h>
> #include [...]

Great. 

Can we get a 2-3 video with same logic instead of photo.?!

FBMinis, 2023-10-26:
I would adapt the project by jameszah to include something like this:

```
static unsigned long lastFrameTime = 0;
unsigned long currentFrameTime = millis();

unsigned long lastPictureTime = 0;
const long interval = 30000; // wait 30sec between sending a new photo+video when there is movement

void loop() {

  // Capture frame every 100ms
  if (currentFrameTime - lastFrameTime >= 100) {

    camera_fb_t* fb = esp_camera_fb_get();

    if (fb) {
      haveMotion = checkMotion(fb, haveMotion);

      if (haveMotion) {
        haveMotion = false;

        Serial.println("movement detected");

        unsigned long currentTime = millis();

        if (currentTime - lastPictureTime >= interval) {
          lastPictureTime = currentTime;

          // capture and send photo+video
          bot.longPoll = 0;
          xTaskCreatePinnedToCore( the_camera_loop, "the_camera_loop", 10000, NULL, 1, &the_camera_loop_task, 1);

        }
      }

      esp_camera_fb_return(fb);
    } else Serial.println("Failed to get frame");

    lastFrameTime = currentFrameTime;
  }

}
```

s60sc, 2023-10-26:
Noted. I've started work on a telegram feature, but will be writing my own code as there's no challenge copying someone else.

FBMinis, 2023-10-26:
One thing I haven't found in other Telegram libs: save error messages in RTC (errors from camera or connections), send them on a Telegram message after the camera has power-cycled or restarted.

s60sc, 2023-10-26:
I'd previously thought about using RTC_NOINIT_ATTR for carrying over restart messages then forgot about it, so I'll look again. Only works for s/w reset not power cycle. The sd logs are persistent though.

s60sc, 2023-11-18:
Telegram & RTC log added to v9.1

closed 2023-11-18 by s60sc