s60sc/ESP32-CAM_MJPEG2SD #297: Core 1 panic'ed on esp32-cam

YummyFatPeanut, 2023-11-03:
On pressing the record button in station mode, the camera (OV2640) immediately resets with a core panic error. 
`16:52:33.482 -> [12:52:33.475 processFrame] Capture started by Button 
16:52:33.515 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.`
Using a 64 gb sd card, the camera saves all configs and logs. ResetReason is-
`12:52:35.157 printResetReason] Software reset due to exception/panic
[12:52:35.157 printWakeupReason] Wakeup by reset`

s60sc, 2023-11-04:
Need to [decode stack trace](https://github.com/s60sc/EspCrashDecoder)
try a smaller card, only cards up to 32gb supported by mmc library

YummyFatPeanut, 2023-11-07:
I have tried an 8 gb card, didn't work.
Now trying to decode trace

closed 2023-11-18 by s60sc