s60sc/ESP32-CAM_MJPEG2SD #298: Startup Failure: Camera XIAO_ESP32S3  init error 0x105

hdruoff, 2023-11-04:
Hi, I am using the XIAO_ESP32S3 board-cam.
The compilation is error-free. The following messages appear at startup:

00:00:00.634 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_E0A9D77554DC started, use 'http://192.168.4.1' to connect
[00:00:04.043 ERROR @ utils.cpp:309] Request failed, error: connection refused 
[00:00:04.043 ERROR @ utils.cpp:323] No External IP returned 
[00:00:04.044 startWifi] Remote server certificates not checked
[00:00:04.044 startPing] Started ping monitoring - On
[00:00:04.046 startWebServer] Starting web server on port: 80
[00:00:04.047 ERROR @ XIAO_ESP32S3-CAM_MJPEG2SD.ino:138] Startup Failure: Camera init error 0x105 
[00:00:06.792 onWiFiEvent] WiFi AP client connection
[00:00:09.045 resetWatchDog] WatchDog started using task: ping
[00:00:00.037 ERROR @ utilsFS.cpp:63] SD card mount failed 
[00:00:00.037 loadConfig] Load config
[00:00:00.038 ERROR @ prefs.cpp:141] Failed to load file /data/configs.txt 
[00:00:00.038 WARN retrieveConfigVal] Key hostName not set 
[00:00:00.039 WARN retE (361) gpio: gpio_set_level(226): GPIO output gpio_num error
[00:00:00.615 setWifiSTA] Wifi Station IP from DHCP
[00:00:00.625 WARN startWifi] SSID  not available, use AP 
[00:00:00.626 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_E0A9D77554DC

Def. Camera:
#include "esp_camera.h"
#include <WiFi.h>
#define CAMERA_MODEL_XIAO_ESP32S3 // Has PSRAM
#include "camera_pins.h"

What is wrong?

s60sc, 2023-11-05:
Follow instructions in appGlobals.h, modify

```
// User's ESP32S3 cam board
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
#endif
```

hdruoff, 2023-11-09:
Thanks, this works for me.
// User's ESP32 cam board
#if defined(CONFIG_IDF_TARGET_ESP32)
#define CAMERA_MODEL_AI_THINKER 

// User's ESP32S3 cam board
//#elif defined(CONFIG_IDF_TARGET_ESP32S3)
//#define CAMERA_MODEL_FREENOVE_ESP32S3_CAM
//#endif
// User's ESP32S3_XIAO cam board
#elif defined(CONFIG_IDF_TARGET_ESP32S3)
#define CAMERA_MODEL_XIAO_ESP32S3
#endif

DJprasenjit, 2023-11-15:
what is the external ip is for? can i connect that ip outside of my local network?
i tried to connect that ip but its not connecting. what else i need to do?

closed 2023-11-18 by s60sc