s60sc/ESP32-CAM_MJPEG2SD #302: Can't start stream on ESP32 AI_Thinker_CAM module

Minha-D, 2023-11-19:
Hi, First of all,Thanks for creating such a amazing project.
I have a problem with starting camera stream on browser. When Start stream is clicked video window remains black. Capturing sills work and they show the preview. And recording on sdcard is also working. But live stream isn't working and sometimes it says something like "not enough free space with error code 500"
I'm using version 9.1 on esp32_cam ai_thinker_board. Any help is appreciated.

s60sc, 2023-11-19:
Need to see log and exact message - is it `500: No free task`

Minha-D, 2023-11-19:
> Need to see log and exact message - is it `500: No free task`

Yes is it "500: No Free Space".
But version 8.7 is working.

Minha-D, 2023-11-19:
I'll attach logs then, to verify what's the problem.

Minha-D, 2023-11-19:
Here is the log from sdcard : [log.txt](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/files/13403299/log.txt)
Here is the screen short of my browser  [image]
[image]

s60sc, 2023-11-19:
No idea. Check you have no other browser instances open, [try chrome browser](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=Browser%20functions%20only%20tested%20on%20Chrome.), or use an older release.

Minha-D, 2023-11-19:
I maked sure no other browsers are open. And also if you looked at the screenshot I'm using Google Chrome only. Thanks for the response I'll use older 8.7 version for now.

closed 2023-11-19 by Minha-D