s60sc/ESP32-CAM_MJPEG2SD #303: Unable to open stream on browser

TNGop, 2023-11-19:
[image]

I have same like Minha-D but in my case it does not show any log like 500: No free task . In my case it shows (ERROR @ utils.cpp:347] Timed out connecting to server: api.ipify.org, Err: ERROR Generic error)
You can check screenshot 
[image]

[image]

s60sc, 2023-11-19:
that error can be ignored as not connected to internet

TNGop, 2023-11-19:
What about starting stream on browser

s60sc, 2023-11-19:
No idea, try an older release.

closed 2023-11-20 by TNGop