s60sc/ESP32-CAM_MJPEG2SD #307: SD Card mount failed

KuntalDa, 2023-11-22:
ESP-ROM:esp32s3-20210327
[00:04:09.214 ERROR @ utilsFS.cpp:63] SD card mount failed 
[00:04:09.214 loadConfig] Load config
[00:04:09.214 ERROR @ prefs.cpp:135] Failed to load file /data/configs.txt 
[00:04:09.214 WARN retrieveConfigVal] Key hostName not set 
[00:04:09.215 WARN ret[00:04:09.622 setupMdnsHost] mDNS service: http://ESP-CAM_MJPEG_F.local
[00:04:13.033 startPing] Started ping monitoring - On
[00:04:13.035 startWebServer] Starting web server on port: 80
[00:04:13.035 ERROR @ ESP32-CAM_MJPEG2SD.ino:138] Startup Failure: Check SD card inserted 
[00:04:13.104 onWiFiEvent] WiFi AP client connection
[00:04:18.033 resetWatchDog] WatchDog started using task: ping

s60sc, 2023-11-22:
user / hardware problem

closed 2023-11-24 by s60sc

FBMinis, 2024-01-15:
I'm having the same problem. The sdcard works properly when running SDMMC_test sketch in the Arduino IDE.

I'm using th AI-Thinker board.

`Used: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi
 Not used: E:\arduino-1.8.19\libraries\WiFi
Using library ESPmDNS at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\ESPmDNS 
Using library Preferences at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\Preferences 
Using library SD_MMC at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\SD_MMC 
Using library FS at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\FS 
Using library LittleFS at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\LittleFS 
Using library Update at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\Update 
Using library WiFi at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi 
Using library HTTPClient at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\HTTPClient 
Using library WiFiClientSecure at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFiClientSecure 
Using library Wire at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\Wire 
"E:\\arduino-1.8.19\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_4751/ESP32-CAM_MJPEG2SD.ino.elf"
`