s60sc/ESP32-CAM_MJPEG2SD #309: Username and password/persisting through Upload, NVS c;ear esptool

EEtinkerer, 2023-11-23:
#
[image]
so I set up the sketch on ESP32-S using a youtube video, instructable.. they were all the same. and it worked great. 
Then beer told me it was a good idea to start playing with settings. 
Needless to say this password screen started coming up after the first reset and I have no clue what the uname and pwrod are. Ive tried uploading sketch over old, clear flash mem upload,  ESPTool flash, write blank sketch, clear NVS sketch, delete MJPEG2SD install different version, format SD card, new SD card.... eveything but a new ESP as been attempted.
this uname and pwrd wont go away

closed 2023-11-24 by s60sc

EEtinkerer, 2023-11-25:
It is definitely occurring, I worked my way down the version list and EVERY version is the same thing.
The only modification I make to the code is as described in the YouTube video, I make sure the camera model AI THINKER is uncommented in appglobals and I add my ssid/pword to utils, long upload to ESP32-S, unplug it, remove SD card, quick format SD card FAT32 default allocation size., insert SD card, plug in to PWR only micro usb cable. connect to camera AP ssid, and navigate to 192.168.4.1.... Same thing every time, I even made a 10minute video, recorded the whole process.
Where does it save the Uname and Pword? I have "Erase all Flash before Upload" enabled, I have uploaded the NVS erase script numerous times.  Is there another secret squirrel location that isnt in these places or on the SD card?
or
Im still learning, but I did find in webserver.cpp ln103 where the Credential/Authentication is checked, can I modify this to get past it? set a Uname & Pword I will remember.

s60sc, 2023-11-25:
Don't change the code, some parts of vid are outdated. Follow readme and do all config via web page.

EEtinkerer, 2023-11-25:
downloaded 9.1, installed, same result

edited webserver.cpp
commented out line 114 (if (strlen(Auth_Name)) {)
all the way to line 132 and the closing bracket for that if statement.

uploaded it, connected to ssid, navigated to 192.168.4.1 and loaded directly in to the WebGUI.
Access settings page still had settings from previous install... setting were not deleted with flash or nvs flash
[image]