s60sc/ESP32-CAM_MJPEG2SD #313: Feature Request: Upload via HTTP Post

thatSFguy, 2023-11-26:
First off, thank you very much for putting this wonderful piece of software together.  Truly impressive how much punch you packed into the small ESP-Cam.  

My request is to have a toggle on the FTP section, so a user could switch to HTTPS and have the file posted to a web server instead.  This way a web-app could be developed to view multiple ESP-Cam recordings in one place, on the web, without having to run a FTP server on the local network. 

a sample upload page could be:
`[code]
<?php
session_start();
$message = ''; 
if (isset($_POST['uploadBtn']) && $_POST['uploadBtn'] == 'Upload')
{
  if (isset($_FILES['uploadedFile']) && $_FILES['uploadedFile']['error'] === UPLOAD_ERR_OK)
  {
    // get details of the uploaded file
    $fileTmpPath = $_FILES['uploadedFile']['tmp_name'];
    $fileName = $_FILES['uploadedFile']['name'];
    $fileSize = $_FILES['uploadedFile']['size'];
    $fileType = $_FILES['uploadedFile']['type'];
    $fileNameCmps = explode(".", $fileName);
    $fileExtension = strtolower(end($fileNameCmps));

    // sanitize file-name
    $newFileName = md5(time() . $fileName) . '.' . $fileExtension;

    // check if file has one of the following extensions
    $allowedfileExtensions = array('jpg', 'gif', 'png', 'zip', 'txt', 'xls', 'doc');

    if (in_array($fileExtension, $allowedfileExtensions))
    {
      // directory in which the uploaded file will be moved
      $uploadFileDir = './uploads/';
      $dest_path = $uploadFileDir . $newFileName;
      if(move_uploaded_file($fileTmpPath, $dest_path)) 
      {
        $message ='File is successfully uploaded.';
      }
      else 
      {
        $message = 'There was some error moving the file to upload directory. Please make sure the upload directory is writable by web server.';
      }
    }
    else
    {
      $message = 'Upload failed. Allowed file types: ' . implode(',', $allowedfileExtensions);
    }
  }
  else
  {
    $message = 'There is some error in the file upload. Please check the following error.<br>';
    $message .= 'Error:' . $_FILES['uploadedFile']['error'];
  }
}
$_SESSION['message'] = $message;
header("Location: index.php");
[/code]`
Of course this would require some authentication added to it, which could just be as easy as random string in a post variable.
The people that would use this feature would also have to get the certificate fingerprint via a web browser, so the wifiClientSecure could be used (https://github.com/hugethor/WiFiClientSecure).

Thanks for your consideration!

s60sc, 2023-11-26:
Thanks
Should be straightforward, but due to time contraints it may take a couple of weeks.

thatSFguy, 2023-11-26:
totally understand the time constraint part!  Looking forward it and it gives me some time to figure out the web interface. Thanks!

closed 2023-12-15 by s60sc

thatSFguy, 2023-12-19:
Thanks!  I'm still under water on a work and personal house project and I can't wait on starting this project...  Many Thanks!