s60sc/ESP32-CAM_MJPEG2SD #315: can't get the IP Address back

ActivePro1, 2023-11-29:
I keep getting this error and I think it maybe my Virgin Router Hub 3 as it worked fine before I changed my internet provider.

ERROR @ utils.cpp:309] Request failed, error: connection refused 
 ERROR @ utils.cpp:323] No External IP returned 

I can change to the temp internet on the ESP32s3 Cam Dev board and enter the IP shown in the serial monitor but this only works once and because I don't have a returned IP Address for my device I can't set up Port Forwarding to see if that helps

Any idea's please as I have tried for days now to sort this issue out

s60sc, 2023-11-29:
[see readme](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=To%20obtain%20your_router_external_ip%20value%2C%20use%20eg%3A%20https%3A//api.ipify.org)

ActivePro1, 2023-11-29:
I have set up Portfording as per your example but 
i still keep get this error
ERROR @ utils.cpp:309] Request failed, error: connection refused
ERROR @ utils.cpp:323] No External IP returned

I have tried and tried again, is there anythink else I can do please

s60sc, 2023-11-29:
it just tells you if your external ip has changed, it doesn't affect the port forwarding itsef

s60sc, 2023-11-29:
you said it worked before, there's nothing I can do, its not a bug. The webcam example doesnt try to get external ip. It doesnt affect the operation of the app.

closed 2023-11-29 by s60sc