s60sc/ESP32-CAM_MJPEG2SD #320: Cannot view on Windows 10 Chrome browser

YapCK45, 2023-12-09:
Compiled and downloaded v9.2 onto ESP32 and everything was a breeze. Opened browser to view but no video. I know it is working because I could download the recorded video. Did I miss out something? Would appreciate comments from anyone. 
Screeen shot below:
[image]

s60sc, 2023-12-09:
streaming takes a lot of bandwidth, your wifi signal strength is low. Use chrome inspector ctrl-shift-i to check connection

YapCK45, 2023-12-10:
Using chrome inspector and checking on the network shows that there is sufficient bandwidth and content is loaded. But there is no display. Please see screen shot.
[image]

YapCK45, 2023-12-10:
I also noticed that the time is not correct. Below screen shot of camera with the system superimposed (in red).
 
[image]

s60sc, 2023-12-10:
The first sustain/stream=0 request is a HEAD request to check a task is available, but the browser is not sending the follow on GET. Also the repeated clockUTC requests indicate a discrepancy in the time sync

maverick-42, 2023-12-10:
> Compiled and downloaded v9.2 onto ESP32 and everything was a breeze. Opened browser to view but no video. I know it is working because I could download the recorded video. Did I miss out something?
> 
> Screeen shot below: ![image](https://private-user-images.githubusercontent.com/64215797/289260666-394aec20-9128-448e-a2d5-c828e909b526.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIyMjk2NTksIm5iZiI6MTcwMjIyOTM1OSwicGF0aCI6Ii82NDIxNTc5Ny8yODkyNjA2NjYtMzk0YWVjMjAtOTEyOC00NDhlLWEyZDUtYzgyOGU5MDliNTI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEwVDE3MjkxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ1NDU2NTgwY2E1NGYxZmM3MjkxYjc4MGRiN2M5M2U5OGY4MzdlNDRkYzA5MGJkNjYxYmIzNzA5NDU2Yjg2MGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MWMl8nHZppEQ5MYfM3IKiWeUyIMg1p-gUh2ZslKOmgw)
> 
> I would like to add that the playback is also working fine. Works more reliably than previous verions. Seems like the problem is not able to see what is streamed. Looked at some of the past comments and seems like some of you encountered this too. Would appreciate comments and suggestions from anyone. It's a long shot unless we are familiar with the code.

I am having the same issue with my board too. Works ok with other codes, but having problem with this one. Motion sensor and Recording stuffs works, but I can't see the stream, or get any still pictures. 

I get this error on my serial monitor, if I press the get still button - ERROR @ appSpecific.cpp:224] Failed to get still.
And if I start stream and then stop it, I get this - [17:28:38.815 showStream] MJPEG: 0 frames, total 0 bytes in 134.4s @ 0.0fps

YapCK45, 2023-12-11:
> The first sustain/stream=0 request is a HEAD request to check a task is available, but the browser is not sending the follow on GET. Also the repeated clockUTC requests indicate a discrepancy in the time sync
> 
> Open console tab for any errors reported

I checked console and this is the info:

Not allowed to load local resource: file:///C:/MyStuff/Arduino/CommonMaster/Latest/common.js

s60sc, 2023-12-11:
That is expected. What matters is any errors when the stream button is pressed

YapCK45, 2023-12-13:
I checked again and there are no errors when the Start Stream button is pressed.

YapCK45, 2023-12-13:
I narrowed down the 'error' in MJPEG2SD.htm. Line 2256 is trying to access a local file which is not allowed for security reason.

[image]

s60sc, 2023-12-13:
As previously stated that error is irrelevant

YapCK45, 2023-12-13:
I am trying to debug the code using VSC. Looks like the coding had been restructured to make use of script files. And common.js is trying to access an array which reference an absolute path to a local resource. That is where the console flags an error on Chrome Inspector that it is attempting to load from a local resource which is not allowed

s60sc, 2023-12-13:
it has nothing to do with stream issue
the issue occurs here:
```
        if (await checkTask('/sustain?stream=0')) {
---->     view.src = webServer + '/sustain?stream=0';
          showView(false);
```
which populates (or doesnt):
`<img id="stream" src="" crossorigin>`

YapCK45, 2023-12-13:
Noted with thanks. I think you can close the issue.

YapCK45, 2023-12-19:
Updated to v9.3 from v8.8 and it works ie it can display the streamed video. However, it has error when motion detect is enable. I have opened another issue for that under #326.

Downloaded the code on **another** ESP32Cam chip (running v8.8) and it doesn't work. I don't think it is the hardware as both chip could work with v8.8.

DJprasenjit, 2023-12-28:
i am using 9.1 version but can not stream shows 500: No free task. but get still is ok but then freeze cant close the still image. manual recording is ok but cant play here

RafikGerges, 2024-01-02:
Hi, After some intensive tracing, the below "if" condition under "processFrame()" never get "true", which results to not giving semaphore, then on the stream server cpp , semaphore can't be taken under "showStream" , we never get to send the stream chunks as the loop continues and continues...@s60sc what do you think?

[image]

RafikGerges, 2024-01-02:
Hi, After some intensive tracing, the below "if" condition under "processFrame()" never get "true", which results to not giving semaphore, then on the stream server cpp , semaphore can't be taken under "showStream" , we never get to send the stream chunks as the loop continues and continues...@s60sc what do you think ?

[image]

s60sc, 2024-01-02:
Thanks, it gives me something to investigate.

s60sc, 2024-01-02:
Uploaded a change to `streamServer.cpp`, but as I dont have the issue I dont know if it makes a difference.

YapCK45, 2024-01-03:
Tested the modified code. Apparently no difference.

RafikGerges, 2024-01-03:
> Uploaded a change to `streamServer.cpp`, but as I dont have the issue I dont know if it makes a difference.

Thank you ! Will do my test as well with some added logging messages for tracing and let you know.

RafikGerges, 2024-01-03:
Also taking a quick look on the changed code, i still think the delay matters, as the semaphore is not given as a result of not meeting the "if" condition on the other cpp.
I might be able to solve it if i know the purpose of that if condition for the buffer size and buffer content( which is always empty), as the malloc will just assign a space in the memory , if you can kindly explain :)

s60sc, 2024-01-03:
processFrame() only fills the buffer if it is empty then sets the semaphore. showStream() sends the buffer when the semaphore is set then clears the buffer size.
I think it is unlikely to be an app issue, browser configuration is more [likely](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/320#issuecomment-1848916924)

YapCK45, 2024-01-06:
I bought a new ESP32 module. Downloaded the code and it works. Motion detection not working as highlighted in Issue #326. Seems like a combination of issues. Need to isolate the root cause.

closed 2024-01-24 by s60sc