s60sc/ESP32-CAM_MJPEG2SD #325: Black screen and telegram not working

TNGop, 2023-12-15:
Your hard work and dedication have unlocked a world of possibilities for this versatile little device. Thanks to your code, ESP32 cams can now be used for everything from capturing stunning wildlife footage to monitoring home security.

I am facing some issues in me esp 32 cam 
My esp 32 get still option is not working 

21:03:08.203 ERROR @ appSpecific.cpp:224] Failed to get still

When I press start stream  it shows black screen

[image]

Telegram integration is not working

[20:50:46.368 ERROR @ utils.cpp:356] Timed out connecting to server: api.telegram.org, Err: ��?��?�$��?#
[20:50:56.568 ERROR @ telegram.cpp:210] Failed to communicate with Telegram server
[20:50:56.569 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[20:50:56.579 checkMemory] Free: heap 103476, block: 77812, min: 97964, pSRAM 441067
[20:50:56.590 loop] =============== Total tasks: 16 ===============

[20:51:04.141 setAlarm] Alarm scheduled at 16/12/2023 01:00:00
[20:51:16.141 ERROR @ utils.cpp:356] Timed out connecting to server: api.ipify.org, Err: �YA?��?�hI�?
[20:51:16.142 doAppPing] Daily rollover
[20:55:34.138 isNight] Night time
[21:00:47.902 wsHandler] Websocket connection: 54
[21:03:08.203 ERROR @ appSpecific.cpp:224] Failed to get still
[image]
[image]

https://github.com/s60sc/ESP32-CAM_MJPEG2SD/assets/102600195/fb4c428d-6ee5-428e-b0d7-7fb19492efe5

ggriesse, 2023-12-19:
I have the same issue as above - that said - Motion detection and recording are working as I can download and watch files created - I can also view the captured motion videos on the Web interfaces - but the live stream is blank and Stills give the same error as above - ERROR @ appSpecific.cpp:223] Failed to get still

s60sc, 2023-12-19:
as I cant recreate the issue I cant investigate. Only someone with the issue can investigate

Rishinet, 2023-12-20:
[14:17:01.584 ERROR @ utils.cpp:356] Timed out connecting to server: api.telegram.org, Err: PK - Memory allocation failed 
[14:17:11.784 ERROR @ telegram.cpp:210] Failed to communicate with Telegram server

h3xx3r, 2023-12-20:
I have the same problem, black screen I think it can't read the file in the data folder correctly and telegram shows ssl certificat error

closed 2023-12-25 by TNGop