s60sc/ESP32-CAM_MJPEG2SD #326: Able to display video after replacing v8.8 with v9.3 but error in motion detection

YapCK45, 2023-12-19:
Downloaded v9.3 and replaced the ESPCam with v8.8 and surprisingly, it is able to display the video. Error message when motion detect is enabled:'

[03:34:46.313 updateAppStatus] Enabling motion detection
E (570579) esp_jpg_decode: JPG Decompression Failed! Parameter error
[03:34:46.454 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed

s60sc, 2023-12-19:
bad frame from cam - it happens, only an issue if it happens a lot

YapCK45, 2023-12-20:
It happens a lot and stopped only when I disable motion detect. See log below:
[12:27:23.165 updateAppStatus] Enabling motion detection
E (77239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:23.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (77439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:23.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (77639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:23.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (77839) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:23.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (78039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:24.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (78239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:24.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (78439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:24.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (78639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:24.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (78840) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:24.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (79039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:25.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (79239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:25.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (79439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:25.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (79639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:25.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (79839) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:25.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (80039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:26.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (80239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:26.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (80439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:26.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (80639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:26.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (80839) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:26.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (81039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:27.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (81239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:27.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (81439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:27.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (81639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:27.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (81839) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:27.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (82039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:28.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (82239) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:28.289 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (82439) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:28.489 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (82639) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:28.689 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (82839) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:28.889 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
E (83039) esp_jpg_decode: JPG Decompression Failed! Parameter error
[12:27:29.089 ERROR @ motionDetect.cpp:187] motionDetect: jpg2rgb() failed 
[12:27:29.210 updateAppStatus] Disabling motion detection

YapCK45, 2023-12-20:
Also has the same problem of no display which I mentioned in issue #320. The same hardware was running v8.8 without any problem. So cannot be hardware.

s60sc, 2023-12-20:
jpeg decoding only occurs during motion detection, and the errors are bad frames from a faulty cam

philippedc, 2023-12-21:
Hi, many thanks for sharing this so great project. Now I have some issues:
1- I first installed the last version - v9.3 - and I cannot get any picture on the browser. Motion is working, recording is working, (I do not use telegram). Note that filenames on SDCard are very strange
[image]

2- I then installed the past version - v8.8 - and I still cannot get any picture on the browser. This time with an error message on the browser
[image]

YapCK45, 2023-12-21:
@s60sc : I compared the relevant section for motionDetect.cpp for v9.3 can v8.8 and they looked the same:

  if (!jpg2rgb((uint8_t*)fb->buf, fb->len, &rgb_buf, (jpg_scale_t)scaling)) {
    LOG_ERR("motionDetect: jpg2rgb() failed");
    free(rgb_buf);
    rgb_buf = NULL;
    return motionStatus;
  }

This issue does not affect v8.8. So I don't think it is due to faulty cam.

YapCK45, 2023-12-21:
@philippedc 
I encountered this problem before. I think your SD card was corrupted. @s60sc : Your comment please.

s60sc, 2023-12-21:
When installing a different version make sure it is a full clean install including data files

philippedc, 2023-12-21:
@YapCK45 it is very strange. I've tried with 2 SDCard of 8Go each. Before insert them to the ESP-CAM I formatted them and test a file copy. 
With v9.3
The first 8Go does not generate error message, therefor  it became a raw device so I cannot control its content with Windows file explorer.
The second 8Go is not recognized by the camera, even if all work controls are well performed with Windows 10
I do not understand....
Nevertheless whatever the SDCard inserted, with no SDCard error message, I cannot get any picture from the web browser. Console messages says that recording and motion works fine.

philippedc, 2023-12-21:
Here is the console message at starting withe the "raw device SDCard"
[image]

s60sc, 2023-12-21:
As per release notes, if using an earlier release you can't use downloaded files which cause 404, you must copy them manually from earlier release zip

philippedc, 2023-12-21:
> When installing a different version make sure it is a full clean install including data files

yes that what I've done. 
When there is not console error message whatever the sdcard used, is there a relationship between sdcard and the fact the camera is not remotely viewable ?

philippedc, 2023-12-21:
Just to clarify I've tried several examples from https://github.com/eloquentarduino/EloquentSurveillance
(with the good 8Go SDCard that is not recognized by ESP32-CAM_MJPEG2SD) and streaming and motion capture work fine.

TotallyNotChloe, 2023-12-21:
About the motion detection error, its also happening to me, same message and everything. Seems to be a error related to bigger resolutions, i lowered it and it seems to have stopped.

Weird thing is that it didn't happened on older versions. just on this new one.

also had a lot of boot loops, had to reinstall multiple times until it stopped.

s60sc, 2023-12-21:
camera can output bad frames which prevents the frame being decoded for motion detection. 
camera modules have varying build quality.

YapCK45, 2023-12-22:
I could replicate @TotallyNotChloe's comment that lowering the resolution seems to solve the motion detect issue. I could get up to sxga(1280X1024). Moving to HD would trigger the ERROR @ motionDetect.cpp:187.

Perhaps it could be due to a imperfect sensor, but then it works on v8.8. Not sure how esp_jpg_decode works. Suggest to trigger ERROR @ motionDetect.cpp:187 only if they occur on multiple times in succesion. This will make it more robust.

In summary, using the **same** hardware. The corresponding data folders are used and updated manually. 

version | motionDetect | streamVideo for display |
| -------- | -------- |-------- |
v8.8| OK   | OK   |
v9.3| NOK   | NOK   |

closed 2024-01-10 by s60sc