s60sc/ESP32-CAM_MJPEG2SD #327: 404: not found

Shadowpost, 2023-12-23:
I'm installing my second ESP cam now. The first one works o.k. but the second gives a 404 error.

I can get a still image.
The camera works and records. The files are readable on the SD card and show up in the web interface

When i choose "start stream", "start playback" or "download", the 404 error will pop-up and there is no file displayed or downloaded.

In my opinion, i have uploaded the sketch exactly like the other camera. Tried via the direct wifi connection as well as via the home wifi. Tried 3 different handhelds and 2 different PC's with different browsers.

Please help!

s60sc, 2023-12-23:
if you are not using the latest release you need to manually copy the old release data to the sd card

Shadowpost, 2023-12-23:
Thanks for the quick response.

Downloading the latest version and programming the board with it fixed the issue!

closed 2024-01-10 by s60sc