s60sc/ESP32-CAM_MJPEG2SD #329: Help with recording videos to SD card

MaelStudio, 2024-01-03:
Hello,

I am currently working on a model rocket flight computer using the XIAO ESP32S3 Sense board. My goal is simply to record AVI videos with audio directly to the SD card, without the need for WiFi functionality. I have had a hard time finding existing basic code that achieves this.

I've experimented with the example code from the Seeed Studio Wiki, but it produces corrupt video files (with no audio), which are not readable unless I convert them to another format through an online converter. This is pretty inconvenient and I have playback framerate issues.

As this repository seems to be built around a complex web app, I am having trouble understanding the code. I have already tried removing stuff I don't need but now the project doesn't compile, and I could use some assistance to get a stripped down version of the code with only the basic video capabilities.

I appreciate any help you can provide.

Thank you.

s60sc, 2024-01-03:
Don't need to change any code, just don't use WiFi. Recordings can be started and stopped using a switch on the pir pin. Settings can be changed by manually editing the config.txt file.

MaelStudio, 2024-01-07:
Thank you for your response! After a few days of reading the code carefully and experimenting, I think I am starting to understand it more. For my application, I really need to remove everything that I won't make use of, even just to make the code simpler to understand, so that it will be easier to add onto that.

closed 2024-01-07 by MaelStudio