s60sc/ESP32-CAM_MJPEG2SD #331: The time is wrong

random-strings, 2024-01-05:
I am running the software on xiao sense esp32s3 version of software 9.3.
This issue I also encountered on esp32cam board software version 9.1.
So the issue is that the system time indicated in windows and the time which is indicated by the ESP32-CAM_MJPEG2SD in the control panel is out of sync for an odd number of minutes.
I don't know why this happens, any ideas or similar expiriences?

bellow are logs that show that time was pulled but it is wrong(out of sync) with the hour on all my other devices.

[11:40:37.256 resetWatchDog] WatchDog started using task: ping  
[11:40:37.257 getLocalNTP] Using NTP server: pool.ntp.org  
[13:40:37.257 showLocalTime] Got current time from NTP: 05/01/2024 13:40:37 with tz: <+02>-2

[image]
[image]

gary-justausername, 2024-01-08:
Hi, I use GMT +0 and mostly 18 minutes fast! Very occasionally correct.

random-strings, 2024-01-10:
> Hi, I use GMT +0 and mostly 18 minutes fast! Very occasionally correct. Bizarre the 18 minutes

Are telegram notifications working for you?

gary-justausername, 2024-01-10:
Sorry I've not tested that yet. I've just finished pan/tilt servos on GPIO1 and GPIO3. Are you having problems with the telegram notifications that I might run into?

closed 2024-01-24 by s60sc