s60sc/ESP32-CAM_MJPEG2SD #332: Version Lite

zilhos, 2024-01-07:
Hello, in my country the espcam version with more memory has not yet arrived. We only have the common version. My question is, could a lite version be made that contains only motion detection through images and notification to telegram. it would be great . From already thank you very much

klimentiy23, 2024-01-07:
It would really be nice to have a lightweight version with the main functions: 1 connect via mutte to the home assistant and work as a binary sensor as a motion detector and record on a flash drive. I think nothing else is needed; many will thank you.

klimentiy23, 2024-01-07:
I tried many times to use the camera as a motion sensor, but alas, my code never worked, simply because the camera says that it cannot connect to my mqtt 
mqtt_binary_sensor:

    mqtt:
      - binary_sensor:
              name: "Cam1"
              state_topic: "homeassistant/sensor/ESP-CAM_1/status"
              value_template: >- 
                  {% if value_json.RECORD == 'ON' %}
                    {{'ON'}}
                  {% else %}
                    {{'OFF'}}
                  {% endif %}
              device_class: motion
              qos: 1

s60sc, 2024-01-07:
If by more memory you mean psram, the app cant work without psram,
The app can be used as a lite version by simply not using / configuring unwanted features.
I cant help with debugging mqtt connectivity to home assistant - I've never used it, only mosquitto. The mqtt feature was contributed by @gemi254

gemi254, 2024-01-09:
I confirm this is a bug because of the stars in mqtt_user_Pass..
Can you try it without password?

gemi254, 2024-01-09:
I had to add  CONFIG_MQTT_PROTOCOL_311 on mqtt.cpp top in order to work properly
Check this pool request
https://github.com/s60sc/ESP32-CAM_MJPEG2SD/pull/334

closed 2024-01-10 by s60sc