s60sc/ESP32-CAM_MJPEG2SD #337: Cant connect to web server

soyabeannnn, 2024-01-14:
After compiling the code, ive tried to used the web server but it doesn't seem to be able to connect to it (via laptop and phone), Ive received this error message as follows:

11:36:27.401 -> [00:07:18.395 WARN getLocalNTP] Not yet synced with NTP 
11:36:39.412 -> [00:07:30.426 ERROR @ utils.cpp:358] Timed out connecting to server: raw.githubusercontent.com, Err: Unknown server error

s60sc, 2024-01-14:
you dont have an internet connection. connect to AP first

soyabeannnn, 2024-01-16:
I see, may i know how do you connect to the AP? and it showed another error where there's error for the camera

12:08:19.486 -> [00:00:04.439 ERROR @ ESP32-CAM_MJPEG2SD.ino:134] Startup Failure: Camera init error 0xffffffff

s60sc, 2024-01-16:
faulty cam, or wrong board selected.

soyabeannnn, 2024-01-16:
sorry but how do u connect to the AP ya?

s60sc, 2024-01-17:
see [readme](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=On%20first%20installation,connection%20is%20available.)

closed 2024-01-24 by s60sc