s60sc/ESP32-CAM_MJPEG2SD #339: Fails during camera_fb_t* fb = esp_camera_fb_get();

FBMinis, 2024-01-16:
My AI-Thinker board always fails when trying to run `camera_fb_t* fb = esp_camera_fb_get();`

The sensor is found during the initialization and running a separate sketch I can see I2C communication is working between esp32 and ov2640, but it does not grab a frame. The board runs other sketches normally that dont use the ov2640 sensor. I looked for shorts in the connector, bad solder points, pressed the connectors against  each other, compared capacitor and resistor values near the connector with other boards I have, nothing worked.

Reference: https://easyeda.com/editor#id=5a6b4b9c6cde4d639ac2b8ab2624d384

I tried with a different camera sensor in this board with the same results, fails to grab frame. Your application works fine with other ESP32-Cam I have. 

This is most likely hardware error. Could you add some debug code to try to understand at what point of the communication through the ov2640 driver it fails?

s60sc, 2024-01-16:
Uncomment   `esp_log_level_set("*", ESP_LOG_NONE); // suppress ESP_LOG_ERROR messages` in `utils.cpp `which may reveal lower level errors. 
Most likely you need to check the hardware signals with a logic analyser. You could try [this](https://github.com/EUA/ESP32_LogicAnalyzer) although I havent used it.

s60sc, 2024-01-16:
make sure the sd card has the latest /data folder content

FBMinis, 2024-01-16:
I formatted the sdcard, re-uploaded the sketch and same result.

FBMinis, 2024-01-17:
Well, I tried with other esp32-cam modules and of course your sketch works perfectly (still discovering all functionalities).

Anyway I will leave a feature request: to include an option to make the sensor/esp32 interaction more verbose to allows us to diagnose possible problems when capturing a frame.

Thank you for sharing you work!

closed 2024-01-24 by s60sc