s60sc/ESP32-CAM_MJPEG2SD #340: freenove esp32 wrover board

antarjcor, 2024-01-17:
Hello
I have downloaded your project.
Very interesting.
I had an ESP32S connected through Arduino Uno, but the program did not work there, understanding that it was due to lack of resources.
It initialized the devices, but the server did not start, or it did not go to IP 192.168.4.1.
I have another simpler program loaded on another ESP32, and it does work, but it does not have the features you have.

I read the specifications and saw the performance with the ESP32S3 board from freenove.
I have ordered one that has this reference.

Freenove ESP32-WROVER board
FNK0060

This board does not have a Micro-SD card reader
It connects via USB and not through Arduino Uno.
I have tested it with several programs and it works correctly, including a simple camera web server that is in the Freenove directory.

I have compiled your project, ESP32-CAM_MJPEG2SD, and loaded it but it does not appear on the IP 192.168.1.4 on the web for the initial configuration.

In preferences files:

https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/61#issuecomment-1034928567
https://raw.githubusercontent.com/espressif/arduino-esp32/ghpages/package_esp32_index.json

Partition scheme
8M with tips

Programmer
Esptool

These are the execution data on the serial monitor:

first: 0x3 (SW_RESET), boot: 0x13 (SPI_FAST_FLASH_BOOT)
configuration: 0, SPIWP: 0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load: 0x3fff0030, len: 1344
load: 0x40078000, len: 13964
load: 0x40080400, len: 3600
entry 0x400805f0
ets July 29, 2019 12:21:46

It is generated in a loop.

In the Arduino ide I have tried with the board type
ESP32S3 development module

But it gives this error.
A fatal error occurred: this chip is ESP32, not ESP32-S3. Wrong chip argument?

In the Arduino IDE it has been defined as

"WRover ESP32 Module" board

I have compiled and loaded with these definitions.

#define CAMERA_MODEL_WROVER_KIT
#define CAMERA_MODEL_AI_THINKER
#define CAMERA_MODEL_M5STACK_ESP32CAM

Is the plate valid?

If so, how should it be declared?

thank you so much

Antarjcor
Spain

[image]

s60sc, 2024-01-18:
ESP32-WROVER is not same as [Freenove ESP32S3 Cam](https://www.aliexpress.com/item/1005004960637276.html?spm=a2g0o.productlist.main.3.29222fbbOuHVsK&algo_pvid=ab92dd9e-8158-4e18-8802-9648eb9b71c1&algo_exp_id=ab92dd9e-8158-4e18-8802-9648eb9b71c1-1&pdp_npi=4%40dis%21GBP%219.82%219.82%21%21%2187.29%2187.29%21%40211b815c17055686519755418e9986%2112000031163496708%21sea%21UK%211887864838%21&curPageLogUid=b33UijKf9Tiu&utparam-url=scene%3Asearch%7Cquery_from%3A)

Bootloop is because compiled for wrong board.

ESP32-WROVER is not supported by app as does not have SD card.

antarjcor, 2024-01-18:
Hello
Thank you so much.

So, my mistake is that I should have bought the reference FNK0085 instead of the FNK0060.

Thank you so much.

Antarjcor
Spain

antarjcor, 2024-01-21:
Hello
I have tested with an FNK0085 board and the application works well.
Slow, but I hope it is a matter of parameters that need to be adjusted.
Thank you

closed 2024-01-21 by antarjcor