s60sc/ESP32-CAM_MJPEG2SD #345: no ap mode or not local ip opening

DJprasenjit, 2024-01-25:
i clicked use https and saved and reboot after that . may local static ip not opening or i connect ap mode and open 192.168.4.1 but not opening

s60sc, 2024-01-25:
Start URL with https://

DJprasenjit, 2024-01-26:
https://192.168.0.177 not opening with https also. what to do, how to disable use https

s60sc, 2024-01-26:
Did you create certificates for 192.168.0.177? 
Did you read the instructions?
To disable https delete the configs.txt file.

DJprasenjit, 2024-01-27:
ok, i deleted config now only using http. but telegram bot is not working, error not enough memory, in 9.5

s60sc, 2024-01-27:
I can't fix your memory problems, use an S3

Nurullah6513, 2024-02-01:
Wifi ismi geliyor bağlanıyorum internet erişimi yok diyor ve verdi ip adreste açılmıyor hata alıyorum sürekli yardımcı olur musunuz

closed 2024-02-11 by s60sc