s60sc/ESP32-CAM_MJPEG2SD #346: Migration to 3.0.0 arduino esp32 core based on ESP-IDF 5.1

rjsachse, 2024-01-25:
Hello, I was wondering if there is plans to update this project to the soon to be released arduino esp32 core v3.0.0 based on ESP-IDF 5.1 currently in alpha phase. It would be good to have a separate branch  since it introduces breaking changes. The major upside is the code is cleaner and reduced, with alot of the api's used in this project updated.

Migration guide from version 2.x to 3.x is available [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html).

s60sc, 2024-01-26:
I have done preliminary changes, but won't release until V3 is stable which will not be soon as it is still in alpha.

rjsachse, 2024-02-06:
can close this, thank you for your reply, I am designing an esp32s3 board with i2s microphone & speaker, sd card, oled display, HKL-LD2450 (motion target tracking sensor), bme280, apds9960,  neopixel, ir rx tx and a pmu (power management unit) I was just doing some testing of software before finale design of hardware. found this project and it is absolutely great.  i previously made some hardware for this project [milight hub](https://github.com/sidoh/esp8266_milight_hub) and WLED project and a few other projects based on esphome etc

have breadboard hardware working with esphome 100% but it does not record video with audio. are you planning to change to vs code or staying with arduino ide?

s60sc, 2024-02-06:
vs code can be used as arduino editor, or are you referring to platformio.

rjsachse, 2024-02-07:
Yer, sorry I ment platformio.

closed 2024-02-11 by s60sc