s60sc/ESP32-CAM_MJPEG2SD #352: V.9.4 Can't change Time Lapse On/Off in Brave Browser on iOS17.

MACKADRIS, 2024-01-30:
Firstly, absolutely AWESOME work you people did on this project.

In version 9.4, while using the wi-fi server and IP address [192.168.4.1], 
I open a Brave Browser window on my iPhone 13 (iOS17.x) and 
attempt to turn on/off the Time Lapse feature to no avail.

The web page .css presents the data info bar located at the bottom of the page, 
but a z-index is putting it in front of the video image pane, 
as well as the data bar in front of the settings pop-outs, 
making the Time Lapse slider inaccessible for any changes.

My successful edit of the .html page:
Changed the z-index of #section-footer to -1
Inserted two line breaks (<br><br>) just after the <image> before closing the <div>.

These changes did not affect the view on my PC browser, 
but did put the video pane on top of the bottom data bar, 
and the video pane is now positioned below the settings popouts for camera, motion, files, etc.

Would changing a 'float' setting somewhere do the same thing, 
creating a space between the video image pane and the bottom data bar?  
The image pane is still over the data bar with my change, and I'm no pro at .css.  
At least, I can now change the Time Lapse On/Off, which was my major hurdle.

I hope this solves someone else's issue trying to go Time Lapse on the fly with an iPhone.  I have not tested this on Android.

Thanks again for this outstanding project  !!
Rick

s60sc, 2024-01-30:
Thanks.
Added 2 breaks to end of motion-record nav to make it scroll above footer, although on chrome can click thru transparent footer.
I dont have access to an IOS device so cant propose any changes beyond yours,

MACKADRIS, 2024-01-30:
Thanks for the quick response and all the great work.
If you make the z-index of #section-footer = -1 , there will be no issue. It’s just moved to the back. I didn’t want to start tearing into all the other z-index and mess up a Java function. 😁

Again, great work. Thanks.

closed 2024-02-11 by s60sc