s60sc/ESP32-CAM_MJPEG2SD #354: [00:00:04.638 ERROR @ ESP32-CAM_MJPEG2SD.ino:134] Startup Failure: Camera init error 0xffffffff

Nurullah6513, 2024-02-02:
[00:00:04.638 ERROR @ ESP32-CAM_MJPEG2SD.ino:134] Startup Failure: Camera init error 0xffffffff
Bu hatayı alıyorum yardımlarınızı bekliyorum teşekkürler

s60sc, 2024-02-02:
hardware problem or compiled for wrong board

Nurullah6513, 2024-02-03:
> Peki nasıl yapmam lazım kısaca nasıl yapıcağımı anlatırsanız çok sevinirim
> teşekkürler
>
> 2 Şub 2024 Cum 18:05 tarihinde s60sc ***@***.***> şunu
> yazdı:

Lütfen yardımcı olun lütfen

2 Şub 2024 Cum 18:09 tarihinde ilahi kervanı ***@***.***> şunu
yazdı:

s60sc, 2024-02-03:
see [readme](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=This%20is%20a,or%20enhancement.%20Thanks.)

Nurullah6513, 2024-02-03:
Yapıyorum yine olmuyor

Nurullah6513, 2024-02-03:
I have been following your projects for a long time, it is very successful.
I have been working on ESP Cam for months. I don't know much about it, but
this is the project that I wanted. Thank you. It is really good. However, I
could not do it. I always get the camera not initialized error. I paid
attention to the appGlobals.h file in your video. The codes there and the
codes in githup are the same, I ask you to please help me with this, thank
you very much in advance, at least maybe you can send me this file
appGlobals.h

s60sc ***@***.***>, 3 Şub 2024 Cmt, 14:37 tarihinde şunu
yazdı:

s60sc, 2024-02-03:
Try a different camera module

nelsonvarela, 2024-02-08:
I have the same issue with two ESP32 camera module's. I bought them from 2 different stores.

```
[00:00:00.271 logSetup] =============== ESP-CAM_MJPEG 9.5 ===============
[00:00:00.282 printResetReason] Power on reset
[00:00:00.282 printWakeupReason] Wakeup by reset
[00:00:00.369 infoSD] SD card type SDHC, Size: 14.8GB
[00:00:00.370 listFolder] Sketch size 1.5MB
[00:00:00.714 listFolder] File: /data/configs.txt, size: 5389 bytes
[00:00:00.714 listFolder] SD_MMC: 6656 bytes used
[00:00:00.715 loadConfig] Load config
[00:00:00.969 updateAppStatus] Enabling motion detection
[00:00:01.823 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.841 WARN startWifi] SSID  not available, use AP 
[00:00:01.843 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_34365965B7A0
[00:00:01.844 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_34365965B7A0 started, use 'http://192.168.4.1' to connect
[00:00:05.858 startPing] Started ping monitoring - On
[00:00:05.862 startWebServer] Starting web server on port: 80
[00:00:05.862 startWebServer] Remote server certificates not checked
[00:00:05.873 ERROR @ ESP32-CAM_MJPEG2SD.ino:134] Startup Failure: Camera init error 0xffffffff 
[00:00:05.873 loop] =============== Total tasks: 12 ===============

[00:00:10.859 resetWatchDog] WatchDog started using task: ping
[00:00:27.262 onWiFiEvent] WiFi AP client connection
[00:01:20.438 onWiFiEvent] WiFi AP client disconnection
```

s60sc, 2024-02-08:
see #363

Nurullah6513, 2024-02-08:
Şimdi kurulum yaptım ama kamera görüntüsü gelmedi ilk baglantida geliyor WiFi SSD bilgilerini girdim ve tekrar bağlantı yaptığımda olmuyor görüntü gelmiyor

nelsonvarela, 2024-02-08:
> see #363

Thanks for the quick response. I will check the other issue.

Nurullah6513, 2024-02-08:
Bide servo yani pan tilt bunu nasıl aktif edicem

closed 2024-02-11 by s60sc