s60sc/ESP32-CAM_MJPEG2SD #357: cam not recording

blackboltdota, 2024-02-04:
the camera stopped recording , showing this error in log  
[00:00:00.511 logSetup] =============== ESP-CAM_MJPEG 9.5 ===============
[00:00:00.521 printResetReason] Power on reset
[00:00:00.521 printWakeupReason] Wakeup by reset
[00:00:00.772 infoSD] SD card type SDHC, Size: 14.8GB
[00:00:00.772 listFolder] Sketch size 1.5MB
[00:00:01.074 listFolder] File: /data/configs.txt, size: 5391 bytes
[00:00:01.077 listFolder] File: /data/common.js, size: 28582 bytes
[00:00:01.080 listFolder] File: /data/MJPEG2SD.htm, size: 95KB
[00:00:01.090 listFolder] SD_MMC: 1.4GB used
[00:00:01.091 loadConfig] Load config
[00:00:01.326 updateAppStatus] Enabling motion detection
[00:00:01.328 setupLamp] Setup Lamp Led for ESP32 Cam board
[00:00:01.541 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_AI_THINKER
[00:00:01.816 setWifiSTA] Wifi Station IP from DHCP
..[00:00:02.647 onWiFiEvent] WiFi Station connection to Knicks Network, using hostname: 1405GateCam
[00:00:02.758 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.182' to connect
[00:00:02.911 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_887C4D9EF0C8
[00:00:02.911 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_887C4D9EF0C8 started, use 'http://192.168.4.1' to connect
[00:00:09.355 startWifi] Wifi stats for Knicks Network - signal strength: -65 dBm; Encryption: WPA_WPA2_PSK; channel: 3
[00:00:09.356 startPing] Started ping monitoring - On
[00:00:09.370 startWebServer] Starting web server on port: 80
[00:00:09.370 startWebServer] Remote server certificates not checked
[00:00:09.371 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:09.381 prepUpload] File uploads will use FTP server
[00:00:09.382 setupLamp] Setup Lamp Led for ESP32 Cam board
[00:00:09.392 prepTelegram] Telegram not being used
[00:00:09.423 updateAppStatus] Enabling motion detection
[00:00:09.559 prepRecording] To record new AVI, do one of:
[00:00:09.559 prepRecording] - press Start Recording on web page
[00:00:09.559 prepRecording] - move in front of camera

[00:00:09.570 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[00:00:09.580 checkMemory] Free: heap 66836, block: 40948, min: 64416, pSRAM 1491379
[00:00:09.581 loop] =============== Total tasks: 16 ===============

[00:00:09.794 resetWatchDog] WatchDog started using task: ping
[00:00:09.794 getLocalNTP] Using NTP server: pool.ntp.org
[05:30:09.795 WARN getLocalNTP] Not yet synced with NTP
[11:42:28.821 isNight] Night time
[11:42:53.751 getLocalNTP] Using NTP server: pool.ntp.org
[11:42:53.751 showLocalTime] Got current time from NTP: 04/02/2024 11:42:53 with tz: IST-5:30
[11:43:24.004 setAlarm] Alarm scheduled at 05/02/2024 01:00:00
[11:43:35.198 ERROR @ utils.cpp:356] Timed out connecting to server: api.ipify.org, Err: X509 - Allocation of memory failed
[11:43:35.199 doAppPing] Daily rollover
[13:39:28.988 WARN pingTimeout] Failed to ping gateway, restart wifi ...
[13:39:28.988 setWifiSTA] Wifi Station IP from DHCP
[13:39:28.991 onWiFiEvent] WiFi Station disconnected
.[13:39:29.068 onWiFiEvent] WiFi Station connection to Knicks Network, using hostname: 1405GateCam
[13:39:29.159 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.182' to connect
[13:39:36.151 startWifi] Wifi stats for Knicks Network - signal strength: -68 dBm; Encryption: WPA_WPA2_PSK; channel: 3
[16:25:58.585 WARN pingTimeout] Failed to ping gateway, restart wifi ...
[16:25:58.586 setWifiSTA] Wifi Station IP from DHCP
[16:25:58.589 onWiFiEvent] WiFi Station disconnected
...[16:25:59.675 onWiFiEvent] WiFi Station connection to Knicks Network, using hostname: 1405GateCam
...[16:26:01.238 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.182' to connect
[16:26:08.246 startWifi] Wifi stats for Knicks Network - signal strength: -63 dBm; Encryption: WPA_WPA2_PSK; channel: 3
[17:05:28.586 WARN pingTimeout] Failed to ping gateway, restart wifi ...
[17:05:28.587 setWifiSTA] Wifi Station IP from DHCP
[17:05:28.590 onWiFiEvent] WiFi Station disconnected
.[17:05:28.759 onWiFiEvent] WiFi Station connection to Knicks Network, using hostname: 1405GateCam
[17:05:28.926 onWiFiEvent] Wifi Station IP, use 'http://192.168.0.182' to connect
[17:05:35.550 startWifi] Wifi stats for Knicks Network - signal strength: -65 dBm; Encryption: WPA_WPA2_PSK; channel: 3
[17:28:26.821 isNight] Day time
[17:48:09.376 ERROR @ webServer.cpp:36] Failed to send /data/MJPEG2SD.htm to browser
[17:48:26.951 wsHandler] Websocket connection: 54
[17:48:36.891 appSpecificWebHandler] JPEG: 110329B in 3513ms
[17:48:45.349 setFolderName] Current directory set to /20240204
[17:48:45.355 ERROR @ utilsFS.cpp:211] Failed to open directory /20240204
[17:48:46.617 setFolderName] Current directory set to /20240204
[17:48:46.637 ERROR @ utilsFS.cpp:211] Failed to open directory /20240204
[17:48:47.497 setFolderName] Previous directory set to /20240203
[17:48:47.503 ERROR @ utilsFS.cpp:211] Failed to open directory /20240203
[17:48:49.377 setFolderName] Previous directory set to /20240203
[17:48:49.401 ERROR @ utilsFS.cpp:211] Failed to open directory /20240203

i think no folder made thats why no video saving

s60sc, 2024-02-04:
sd card no longer connected, push it in properly - see [readme](https://github.com/s60sc/ESP32-CAM_MJPEG2SD#:~:text=This%20is%20a,or%20enhancement.%20Thanks.)

blackboltdota, 2024-02-04:
but i am able to access older files

blackboltdota, 2024-02-05:
also the folder does create when i press start recording

closed 2024-02-11 by s60sc