s60sc/ESP32-CAM_MJPEG2SD #358: Camera fails to download /data/config.txt, because it was removed from github repository

StupidProgramming, 2024-02-04:
Hi, 

I get the following error:
[14:53:50.039 wgetFile] Downloading /data/configs.txt from /s60sc/ESP32-CAM_MJPEG2SD/master/data/configs.txt
[14:53:50.286 ERROR @ setupAssist.cpp:39] Download failed, error:
[14:53:50.292 ERROR @ setupAssist.cpp:46] HTTP Get failed with code: 404

and when i check, i see the configs.txt file has been removed form the repository 2 weeks ago.

[image]

Can you please put the file back?

s60sc, 2024-02-04:
either use latest release or get file from older release

StupidProgramming, 2024-02-04:
Okay, but it doesn't hurt to keep older versions working right?

rjsachse, 2024-02-06:
> Okay, but it doesn't hurt to keep older versions working right?

I don't understand the reason to keep older versions working for a non commercial project. best bet if having issues upload to latest version or make a pr with changes

closed 2024-02-11 by s60sc