s60sc/ESP32-CAM_MJPEG2SD #361: Err: RSA - The public key operation failed : BIGNUM - Memory allocation failed

zilhos, 2024-02-06:
Hello, I appreciate your great work, I ask you, is it possible for Telegram to work with this amount of  1489183 pSRAM? Could the sketch be made smaller so that only the motion detection by camera and the notification to telegran work.....? Thanks for your answer
I reduced all functions to a minimum, still I can't get it to connect to Telegram

[17:43:28.932 startPing] Started ping monitoring - Off
[17:43:28.960 startWebServer] Starting web server on port: 80
[17:43:28.960 startWebServer] Remote server certificates not checked
[17:43:28.961 startSustainTasks] Started 1 HTTP sustain tasks
[17:43:28.971 prepUpload] File uploads will use FTP server
[17:43:28.972 resetWatchDog] WatchDog started using task: ping
[17:43:28.982 getLocalNTP] Using NTP server: pool.ntp.org
[17:43:28.982 setupLamp] Setup Lamp Led for ESP32 Cam board
[14:43:28.984 showLocalTime] Got current time from NTP: 06/02/2024 14:43:28 with tz: <-03>3
[14:43:39.828 ERROR @ utils.cpp:356] Timed out connecting to server: api.telegram.org, Err: RSA - The public key operation failed : BIGNUM - Memory allocation failed
[14:43:49.839 ERROR @ telegram.cpp:214] Failed to communicate with Telegram server
[14:43:49.877 updateAppStatus] Disabling motion detection
[14:43:50.035 prepRecording] To record new AVI, do one of:
[14:43:50.036 prepRecording] - press Start Recording on web page

[14:43:50.036 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[14:43:50.047 checkMemory] Free: heap 67664, block: 34804, min: 25124, pSRAM 1489183
[14:43:50.057 loop] =============== Total tasks: 16 ===============

s60sc, 2024-02-06:
Not seen that particular error before, but the problem is dram heap not psram. TLS for secure connections uses a lot of memory.
However I'm using the base functions with telegram successfully on my esp32. 
```
[12:19:29.294 startPing] Started ping monitoring - On
[12:19:29.309 startWebServer] Starting web server on port: 80
[12:19:29.309 startWebServer] Remote server certificates not checked
[12:19:29.310 startSustainTasks] Started 1 HTTP sustain tasks
[12:19:29.320 prepUpload] File uploads will use FTP server
[12:19:29.329 resetWatchDog] WatchDog started using task: ping
[12:19:29.329 getLocalNTP] Using NTP server: pool.ntp.org
[12:19:29.340 showLocalTime] Got current time from NTP: 06/02/2024 12:19:29 with tz: GMT0
[12:19:30.795 prepTelegram] Connected to Telegram Bot Handle: GarageCam
[12:19:30.837 updateAppStatus] Enabling motion detection
[12:19:31.020 prepRecording] To record new AVI, do one of:
[12:19:31.021 prepRecording] - press Start Recording on web page
[12:19:31.021 prepRecording] - move in front of camera

[12:19:31.034 setup] Camera model OV2640 on board CAMERA_MODEL_AI_THINKER ready @ 20MHz
[12:19:31.035 checkMemory] Free: heap 47424, block: 22516, min: 44592, pSRAM 1489527
[12:19:31.047 loop] =============== Total tasks: 17 ===============
```
For some reason your `min:` is much less than mine.
When arduino is updated to idf 5.1 memory management should be better.

zilhos, 2024-02-07:
Thanks for your prompt response. My esp32 is not S3. In my country we don't have S3, I think that's why the difference with your memory.My es32 is only model S
Thank you for your great work

s60sc, 2024-02-07:
The log is from an esp32, same model as yours

closed 2024-02-11 by s60sc