s60sc/ESP32-CAM_MJPEG2SD #363: Camera init error 0xffffffff

InokentiySSL, 2024-02-08:
I bought 2 camera models and get the same error. The standard example from arduino ide works and the camera shows. Why is that?
[image]
[image]
[image]

InokentiySSL, 2024-02-08:
This project is also working.
https://github.com/easytarget/esp32-cam-webserver
[image]
[image]

s60sc, 2024-02-08:
in `ESP32-CAM_MJPEG2SD.ino `replace:

`  if (psramFound()) prepCam();`

with:

```
  if (psramFound()) {
    LOG_INF("PSRAM size: %s", fmtSize(esp_spiram_get_size()));
    prepCam();
  }
```

Need at least 4MB PSRAM

nelsonvarela, 2024-02-08:
I think this issue can be closed. My CAM has 2MB PSRAM but the seller has 4MPSRAM stated on the website.... misinformation!

I can also see in @InokentiySSL's screenshot he has 2095071 bytes which is not enough.

Thanks @s60sc for your patience and willing to help.

s60sc, 2024-02-09:
Thanks for that useful information.  I'll add a note to the readme

closed 2024-02-11 by s60sc

soyabeannnn, 2024-03-30:
But the thing is I've used different boards from different sellers and it still shows the PSRAM error.

The web server doesn't show up but only says Startup Failure: Insufficient PSRAM for app

guece1, 2024-04-13:
I have this problem with my esp32s3 and OV5640

Compilation error: 'esp_spiram_get_size' was not declared in this scope

the problem is on Arduino core for the ESP32 3.0.0-rc1 and 3.0.0-alpha3
on 2.0.15 and 2.0.14 i ca compile but my cam module wont startup
Startup Failure: Camera init error 0xffffffff on CAMERA_MODEL_ESP32S3_EYE_FREENOVE