s60sc/ESP32-CAM_MJPEG2SD #365: Can't compile with arduino-cli

PsySc0rpi0n, 2024-02-10:
Hi.
I'm trying to compile this repository using `ardudino-cli` with no success. I'm on Debian Bookworm.
I'm using an ESP32-CAM (AI TINKER).
https://pt.aliexpress.com/item/1005006074477423.html

Some info:
```
$ arduino-cli version
arduino-cli  Version: 0.35.2 Commit: 01de174c Date: 2024-01-24T11:33:03Z
```
```
$ arduino-cli board list
Port         Protocol Type              Board Name FQBN Core
/dev/ttyUSB0 serial   Serial Port (USB) Unknown
```

```
$ arduino-cli core list
ID          Installed    Latest       Name
esp32:esp32 3.0.0-alpha3 3.0.0-alpha3 esp32
```

Compiling with these options:
```
arduino-cli compile --upload --port /dev/ttyUSB0 --fqbn esp32:esp32:esp32:JTAGAdapter=default,PartitionScheme=min_spiffs,PSRAM=enabled,FlashMode=dio,FlashSize=4M,FlashFreq=80,LoopCore=1,EventsCore=1,CPUFreq=240,UploadSpeed=921600,EraseFlash=all --verbose ~/Arduino/esp32/ESP32-CAM_MJPEG2SD/ESP32-CAM_MJPEG2SD.ino
```

Getting the following output (trimmed):
```
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/motionDetect.cpp: In function 'bool jpg2rgb(const uint8_t*, size_t, uint8_t**, jpg_scale_t)':
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/motionDetect.cpp:375:50: error: invalid conversion from 'uint32_t (*)(void*, size_t, uint8_t*, size_t)' {aka 'long unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} to 'jpg_reader_cb' {aka 'unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)'} [-fpermissive]
  375 |   esp_err_t res = esp_jpg_decode(src_len, scale, _jpg_read, _rgb_write, (void*)&jpeg);
      |                                                  ^~~~~~~~~
      |                                                  |
      |                                                  uint32_t (*)(void*, size_t, uint8_t*, size_t) {aka long unsigned int (*)(void*, unsigned int, unsigned char*, unsigned int)}
In file included from /home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/espressif__esp32-camera/conversions/include/img_converters.h:25,
                 from /home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/espressif__esp32-camera/driver/include/esp_camera.h:250,
                 from /home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/appGlobals.h:176,
                 from /home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/motionDetect.cpp:23:
/home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/espressif__esp32-camera/conversions/include/esp_jpg_decode.h:37:71: note:   initializing argument 3 of 'esp_err_t esp_jpg_decode(size_t, jpg_scale_t, jpg_reader_cb, jpg_writer_cb, void*)'
   37 | esp_err_t esp_jpg_decode(size_t len, jpg_scale_t scale, jpg_reader_cb reader, jpg_writer_cb writer, void * arg);
      |                                                         ~~~~~~~~~~~~~~^~~~~~
/home/psysc0rpi0n/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -MMD -c @/home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/flags/cpp_flags -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_min_spiffs "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PartitionScheme=min_spiffs,PSRAM=enabled,FlashMode=dio,FlashSize=4M,FlashFreq=80,LoopCore=1,EventsCore=1,CPUFreq=240,UploadSpeed=921600,EraseFlash=all\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0 @/home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/flags/defines -I/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD -iprefix /home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/ @/home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/flags/includes -I/home/psysc0rpi0n/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/dio_qspi/include -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32 -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/variants/esp32 -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/ESPmDNS/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/Preferences/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/SD_MMC/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/FS/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/LittleFS/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/Update/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/WiFi/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/HTTPClient/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/WiFiClientSecure/src -I/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/Wire/src @/tmp/arduino/sketches/175F0E299CDAD5F33FAE2B3DE68D9E28/build_opt.h @/tmp/arduino/sketches/175F0E299CDAD5F33FAE2B3DE68D9E28/file_opts /tmp/arduino/sketches/175F0E299CDAD5F33FAE2B3DE68D9E28/sketch/mqtt.cpp -o /tmp/arduino/sketches/175F0E299CDAD5F33FAE2B3DE68D9E28/sketch/mqtt.cpp.o
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp: In function 'void controlFrameTimer(bool)':
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:92:5: error: 'timerAlarmDisable' was not declared in this scope
   92 |     timerAlarmDisable(frameTimer);
      |     ^~~~~~~~~~~~~~~~~
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:98:28: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   98 |     frameTimer = timerBegin(3, 8000, true); // 0.1ms tick
      |                  ~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal.h:84,
                 from /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/Arduino.h:36,
                 from /home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/globals.h:16,
                 from /home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/appGlobals.h:6,
                 from /home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:9:
/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-timer.h:35:14: note: declared here
   35 | hw_timer_t * timerBegin(uint32_t frequency);
      |              ^~~~~~~~~~
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:101:5: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  101 |     timerAlarmWrite(frameTimer, frameInterval, true);
      |     ^~~~~~~~~~~~~~~
      |     timerWrite
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:102:5: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  102 |     timerAlarmEnable(frameTimer);
      |     ^~~~~~~~~~~~~~~~
      |     timerAlarm
/home/psysc0rpi0n/Arduino/esp32/ESP32-CAM_MJPEG2SD/mjpeg2sd.cpp:103:25: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  103 |     timerAttachInterrupt(frameTimer, &frameISR, true);
      |     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
[... 34 lines cut ...]
Using library LittleFS at version 2.0.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/LittleFS 
Using library Update at version 2.0.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/Update 
Using library WiFi at version 2.0.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/WiFi 
[... cut ...]
HTTPClient       2.0.0   /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/HTTPClient
WiFiClientSecure 2.0.0   /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/WiFiClientSecure
Wire             2.0.0   /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/libraries/Wire

Used platform Version      Path
esp32:esp32   3.0.0-alpha3 /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3
Error during build: exit status 1
```

What am I missing?

s60sc, 2024-02-10:
compile with stable arduino core not an alpha

PsySc0rpi0n, 2024-02-10:
> compile with stable arduino core not an alpha

You mean with a stable release of `arduino-cli` ? Or an older version of the ESP32 board? Coul you please clarify?
Thank you
Psy

closed 2024-02-10 by PsySc0rpi0n