s60sc/ESP32-CAM_MJPEG2SD #366: Startup Failure: Insufficient PSRAM for app

EP-man, 2024-02-14:
Hi. i have an ESP32-CAM and it has 4mb psram. I write all the required lines for the code and put them into the code, after writing it on the board and getting a feedback from it, this error shows:
`[00:00:00.302 printResetReason] Power on reset
[00:00:00.302 printWakeupReason] Wakeup by reset
[00:00:00.348 infoSD] SD card type SDHC, Size: 29.5GB
[00:00:00.349 listFolder] Sketch size 1.5MB
[00:00:00.649 listFolder] File: /data/configs.txt, size: 5532 bytes
[00:00:00.650 listFolder] SD_MMC: 588.7MB used
[00:00:00.650 loadConfig] Load config
[00:00:00.894 updateAppStatus] Enabling motion detection
[00:00:00.906 setup] PSRAM size: 2.0MB
[00:00:01.120 setWifiSTA] Wifi Station IP from DHCP
[00:00:01.133 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_90304F67B4FC
[00:00:01.136 WARN startWifi] SSID  not available, use AP 
[00:00:01.137 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_90304F67B4FC started, use 'http://192.168.4.1' to connect
[00:00:03.951 startPing] Started ping monitoring - On
[00:00:03.953 startWebServer] Starting web server on port: 80
[00:00:03.954 startWebServer] Remote server certificates not checked
[00:00:03.964 ERROR @ ESP32-CAM_MJPEG2SD.ino:138] Startup Failure: Insufficient PSRAM for app 
[00:00:03.964 loop] =============== Total tasks: 12 ===============`

s60sc, 2024-02-14:
Your log shows:
`[00:00:00.906 setup] PSRAM size: 2.0MB`
You have a [fake board](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/363#issuecomment-1935037553)

EP-man, 2024-02-15:
Thanks for the reply. Now I have to contact the seller and see what's the problem.

closed 2024-03-03 by s60sc