s60sc/ESP32-CAM_MJPEG2SD #367: Not working on esp32 wrover dev

slavikfoxy, 2024-02-17:
Hello, I set up a sketch, recorded a program and it does not work, my board is esp32 wrover dev

22:09:51.195 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:09:51.195 -> configsip: 0, SPIWP:0xee
22:09:51.195 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:09:51.243 -> mode:DIO, clock div:1
22:09:51.243 -> load:0x3fff0030,len:1184
22:09:51.243 -> load:0x40078000,len:13260
22:09:51.243 -> load:0x40080400,len:3028
22:09:51.243 -> entry 0x400805e4
22:09:52.212 -> 
22:09:52.212 -> 
22:09:52.212 -> [00:00:00.463 logSetup] Setup RAM based log, size 7168, starting from 0
22:09:52.212 -> 
22:09:52.212 -> 
22:09:52.212 -> [00:00:00.464 logSetup] =============== ESP-CAM_MJPEG 9.6 ===============
22:09:52.212 -> [00:00:00.474 printResetReason] Power on reset
22:09:52.212 -> [00:00:00.475 printWakeupReason] Wakeup by reset

s60sc, 2024-02-17:
Inadequate power supply

slavikfoxy, 2024-02-26:
I can't run it, I changed partition scheme, code version, board. my esp32 cam has a sd-reader, I bought this module, if I connect it is there a chance that it will work? 

09:56:56.495 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:56:56.495 -> configsip: 0, SPIWP:0xee
09:56:56.495 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:56:56.495 -> mode:DIO, clock div:1
09:56:56.495 -> load:0x3fff0030,len:1184
09:56:56.495 -> load:0x40078000,len:13260
09:56:56.495 -> load:0x40080400,len:3028
09:56:56.495 -> entry 0x400805e4
09:56:57.466 -> 
09:56:57.466 -> 
09:56:57.466 -> =============== ESP-CAM_MJPEG 9.1.2 ===============
09:56:57.466 -> [00:00:00.464 logSetup] Setup RAM based log, size 7168, starting from 0
09:56:57.512 -> 
09:56:57.512 -> 
09:56:57.512 -> [00:00:00.464 printResetReason] Power on reset
09:56:57.512 -> [00:00:00.475 printWakeupReason] Wakeup by reset

slavikfoxy, 2024-02-26:
It doesn't work, I've tried several different cables, different usb ports, soldered capacitors, and nothing helps. I think this board is defective. But the rest of the code works. Is there any way to skip this error?

[image]
[image]

s60sc, 2024-02-26:
`But the rest of the code works` doesnt make sense. Is the cam bootlooping?

slavikfoxy, 2024-02-26:
No, the camera does not work, but other code, such as AsyncTelegram2, works.

s60sc, 2024-02-26:
have you tried other camera apps?

s60sc, 2024-02-26:
then its a problem with the usb power supply not the board

slavikfoxy, 2024-02-27:
I soldered the usb to the vin and connected it to the laptop and the esp32 burned out, or rather the stabilizer on the springs, it was giving out 5v instead of 3.3v. I tried to connect 3.3v from the esp8266 and realized that the processor was dead. It's strange that I put 5v to the same place as the usb line, but it didn't survive. It looks like my board was defective, I'll try to buy something better.

Translated with DeepL :)
[image]
[image]
[image]
[image]

closed 2024-02-27 by slavikfoxy