s60sc/ESP32-CAM_MJPEG2SD #371: Error UtilsFS.cpp:263

ActivePro2, 2024-02-26:
My camera wil work fine for several days but then I can load up the link ok but when I try to stream I get a blank screen with this error.

110:04:57.483 ERROR @ utilsFS.cpp:263] Failed to open

Any idea’s please on why this happens

ActivePro2, 2024-02-26:
Just sorted this error I think, I looked at the logs and it mentioned the SD card, I took the card out, deleted all the video’s and reinserted the SD card, it’s now working again.

ActivePro2, 2024-02-26:
> if you switch off automatic deletion when full you have to do it manually

That sorted it out, thank you

closed 2024-02-26 by ActivePro2

ActivePro2, 2024-06-04:
I would like to run two of the cameras on the same WI-FI, the code will name the second camera AP the same as the first, will this not be an issue as they will both want to use the same named AP ESP_CAM_MJPEG_18C7A6188534 or is there a way to rename the second camera please.

s60sc, 2024-06-04:
Edit Config / Wifi / AP SSID name