s60sc/ESP32-CAM_MJPEG2SD #376: Apologies and a little help please.

OldYin, 2024-03-01:
Apologies as this is not necessarily an issue with software but I am having issues with the AVI output format.

Everything else pretty much OK just that when I download any of the captured videos to ANY device, the format is not recognised.
I CAN play all videos with FFMPEG and VLC player but Media Player Film & TV (Windows 10) and any default apps on mobile phones and tablets (All Android) won't look at the files.
I can hex any one of the files and change one byte and suddenly I can play the files across all devices.
I don't know enough about the format to understand why, perhaps you can help?

regards,
Alex

s60sc, 2024-03-01:
What is the one byte change?

OldYin, 2024-03-01:
[image]

Changing the last byte to anything other than '00' pretty much ensures that the video is readable by all default windows 10 players / editors. Actually make any of the values in column 5 non zero and that will work. I'm most likely doing nothing useful but it is a little faster than running it through ffmpeg and creating an MP4. 

As for my Android devices; they are not natively able to decode MJPG files. The reason they are now is because in the last couple of days we installed a Smart Doorbell. The app associated with the Doorbell camera has installed a new default player which can decode MJPG. Sorry for that bit of misinformation. 

Regards,
Alex

s60sc, 2024-03-01:
Changing that byte corrupts indexed chunk length but many players such as VLC can fix it on the fly. 
Apart from Media Player Film & TV which I dont have (but Media Player Classic works fine), what other specific apps can play the corrupted file but not the original file.
Android doesnt play AVI files natively, but lots of third party video apps do.

OldYin, 2024-03-01:
Thanks for having a look any way. 

I was poking about a bit doing nothing useful as usual and found that incrementing the idx1 index size also has the same result.

Other than Media Player / Films & TV / Windows Media Player (is that classic?) I don't have much else installed which I use regularly. Movie Maker & AVS video converter also won't play the files.

Interestingly, doing a straight codec copy with ffmpeg results in a MJPG file with is totally playable with all of the windows apps. I'm not really getting any clues from the hex editor though. I guess I'll be installing VLC over everything now :)

Thanks again for looking.

Alex

closed 2024-03-03 by s60sc

OldYin, 2024-03-04:
It appears that idxOffset[isTL] needs a value of 4 offset.
Now .avi plays in all windows apps including VLC player with no errors. 
Made a couple of changes in avi.cpp

regards,
Alex
**************************************
void prepAviIndex(bool isTL) {
  // prep buffer to store index data, gets appended to end of file
  if (idxBuf[isTL] == NULL) idxBuf[isTL] = (uint8_t*)ps_malloc((maxFrames+1)*IDX_ENTRY); // include some space for audio index
  memcpy(idxBuf[isTL], idx1Buf, 4); // index header
  idxPtr[isTL] = CHUNK_HDR;  // leave 4 bytes for index size
  moviSize[isTL] = indexLen[isTL] = 0;
  idxOffset[isTL]=4; //4 byte offset
}

void buildAviHdr(uint8_t FPS, uint8_t frameType, uint16_t frameCnt, bool isTL) {
  // update AVI header template with file specific details
  size_t aviSize = moviSize[isTL] + AVI_HEADER_LEN + ((CHUNK_HDR+IDX_ENTRY) * (frameCnt+(haveSoundFile?1:0))); // AVI content size 
  // update aviHeader with relevant stats
  memcpy(aviHeader+4, &aviSize, 4);
  uint32_t usecs = (uint32_t)round(1000000.0f / FPS); // usecs_per_frame 
  memcpy(aviHeader+0x20, &usecs, 4); 
  memcpy(aviHeader+0x30, &frameCnt, 2);
  memcpy(aviHeader+0x8C, &frameCnt, 2);
  memcpy(aviHeader+0x84, &FPS, 1);
  uint32_t dataSize = moviSize[isTL] + ((frameCnt+(haveSoundFile?1:0)) * CHUNK_HDR) + 4; 
  memcpy(aviHeader+0x12E, &dataSize, 4); // data size 

  // apply video framesize to avi header
  memcpy(aviHeader+0x40, frameSizeData[frameType].frameWidth, 2);
  memcpy(aviHeader+0xA8, frameSizeData[frameType].frameWidth, 2);
  memcpy(aviHeader+0x44, frameSizeData[frameType].frameHeight, 2);
  memcpy(aviHeader+0xAC, frameSizeData[frameType].frameHeight, 2);

#if INCLUDE_MIC
  uint8_t withAudio = 2; // increase number of streams for audio
  if (isTL) memcpy(aviHeader+0x100, zeroBuf, 4); // no audio for timelapse
  else {
    if (haveSoundFile) memcpy(aviHeader+0x38, &withAudio, 1); 
    memcpy(aviHeader+0x100, &audSize, 4); // audio data size
  }
  // apply audio details to avi header
  memcpy(aviHeader+0xF8, &SAMPLE_RATE, 4);
  uint32_t bytesPerSec = SAMPLE_RATE * 2;
  memcpy(aviHeader+0x104, &bytesPerSec, 4); // suggested buffer size
  memcpy(aviHeader+0x11C, &SAMPLE_RATE, 4);
  memcpy(aviHeader+0x120, &bytesPerSec, 4); // bytes per sec
#else
  memcpy(aviHeader+0x100, zeroBuf, 4);
#endif

  // reset state for next recording
  //moviSize[isTL] = idxOffset[isTL] = idxPtr[isTL] = 0;

  moviSize[isTL] = idxPtr[isTL]=0;
  idxOffset[isTL]=4;  //4 byte offset
}
**************************************