s60sc/ESP32-CAM_MJPEG2SD #377: help

dekaydk, 2024-03-01:
after install serial port just shows
motion when camera moves but
nothing saved
no ip address or any other info

s60sc, 2024-03-01:
Provide startup log

dekaydk, 2024-03-01:
cant copy whole log
this is begin

[00:00:00.747 WARN retrieveConfigVal] Key framesize not set 
[00:00:01.015 setWifiSTA] Wifi Station IP from DHCP
.[00:00:01.383 onWiFiEvent] WiFi Station connection to home8402, using hostname: cmst
.[00:00:01.691 onWiFiEvent] Wifi Station IP, use 'http://10.0.0.45' to connect
[00:00:05.563 startWifi] Wifi stats for home8402 - signal strength: -62 dBm; Encryption: WPA2_PSK; channel: 11
[00:00:05.564 startPing] Started ping monitoring - On
[00:00:05.578 startWebServer] Starting web server on port: 80
[00:00:05.578 startWebServer] Remote server certificates not checked
[00:00:05.578 ERROR @ ESP32-CAM_MJPEG

closed 2024-03-01 by dekaydk