s60sc/ESP32-CAM_MJPEG2SD #379: Does not connect to the WiFi network

mamazad7, 2024-03-04:
Environment : 
ESP32-CAM_MJPEG2SD ver 9.6.1
IDE: Arduino IDE ver 2.3.1
CAM board:  S3
workstation: Win 11
Browser: Chrome 

Hi, tried to enter the WiFi SSID/PW via the web page (192.168.4.1) but had no success. Then tried to hard code in "utils.cpp" file but still had no luck. it looks like the PW won't save where it should. I can see the SSID in the config.txt and log file.
Additionally, tried with a test WiFi sketch in the same environment and hardware. had no issues: connected to the same WiFi AP/router with DHCP. 
following is the log file from IDE serial monitor.
Thank you.
_______________________________

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42052a00
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9a8
load:0x403cc700,len:0x2920
entry 0x403c98b8

[00:00:01.641 logSetup] Setup RAM based log, size 7168, starting from 754

[00:00:01.641 logSetup] =============== ESP-CAM_MJPEG 9.6 ===============
[00:00:01.641 printResetReason] Reset for unknown reason
[00:00:01.641 printWakeupReason] Wakeup by reset
[00:00:01.828 infoSD] SD card type SDHC, Size: 7.4GB
[00:00:01.829 listFolder] Sketch size 1.4MB
[00:00:01.992 listFolder] File: /data/common.js, size: 28564 bytes
[00:00:01.996 listFolder] File: /data/MJPEG2SD.htm, size: 95KB
[00:00:01.999 listFolder] File: /data/configs.txt, size: 5534 bytes
[00:00:02.000 listFolder] SD_MMC: 7.5MB used
[00:00:02.001 loadConfig] Load config
[00:00:02.198 updateAppStatus] Enabling motion detection
[00:00:02.202 setup] PSRAM size: 8.0MB
[00:00:02.374 prepCam] Camera init OK for model OV2640 on board CAMERA_MODEL_NEW_ESPS3_RE1_0
[00:00:02.590 setWifiSTA] Wifi Station IP from DHCP
.....[00:00:05.013 onWiFiEvent] WiFi Station disconnected
.....[00:00:07.421 onWiFiEvent] WiFi Station disconnected
[00:00:07.613 WARN startWifi] SSID HomeNetWiFi unable to connect 
[00:00:07.616 onWiFiEvent] Wifi AP stopped: ESP-CAM_MJPEG_D03A85E22748
[00:00:07.617 onWiFiEvent] Wifi AP SSID: ESP-CAM_MJPEG_D03A85E22748 started, use 'http://192.168.4.1' to connect
[00:00:10.826 startWifi] **Wifi stats for HomeNetWiFi - signal strength: -69** dBm; Encryption: WPA2_PSK; channel: 2
[00:00:10.827 startPing] Started ping monitoring - Off
[00:00:10.831 startWebServer] Starting web server on port: 80
[00:00:10.831 startWebServer] Remote server certificates not checked
[00:00:10.832 startSustainTasks] Started 1 HTTP sustain tasks
[00:00:10.832 prepUpload] File uploads will use FTP server
[00:00:10.854 updateAppStatus] Enabling motion detection
[00:00:10.967 prepRecording] To record new AVI, do one of:
[00:00:10.968 prepRecording] - press Start Recording on web page
[00:00:10.968 prepRecording] - move in front of camera
 
[00:00:10.968 setup] Camera model OV2640 on board CAMERA_MODEL_NEW_ESPS3_RE1_0 ready @ 20MHz
[00:00:10.968 checkMemory]  Free: heap 174632, block: 163828, min: 174632, pSRAM 2600127
[00:00:10.969 loop] =============== Total tasks: 16 ===============

[00:00:13.621 isNight] Night time
[00:00:15.828 resetWatchDog] WatchDog started using task: ping
[00:00:15.829 WARN pingTimeout] Disconnected, restart wifi ... 
[00:00:15.829 setWifiSTA] Wifi Station IP from DHCP
.[00:00:16.225 isNight] Day time
..[00:00:17.145 onWiFiEvent] WiFi AP client connection
....[00:00:19.050 onWiFiEvent] WiFi Station disconnected
..[15:17:35.316 showLocalTime] Got current time from browser: 03/03/2024 15:17:35 with tz: <-10>10

s60sc, 2024-03-04:
`[00:00:07.613 WARN startWifi] SSID HomeNetWiFi unable to connect` Probably wrong password
Set password on web page in Access Settings.
Dont modify code.

mamazad7, 2024-03-04:
yes, I tried using the web page ( as attached pic) a few times before modifying the code but no luck. The password is correct as I tried the same SSID/PW ( cut/paste and manually enter) with another WiFi code;  worked fine. 
what I feel is 
a. PW word doesn't save in NVS
b. adding or chopping   characters while saving
c. my PW is alphanumeric and has the symbol #

pls note this is a closed WiFi network that does not have internet access.
[image]

arkimarki, 2024-03-07:
I have the same problem!
[18:12:15.941 WARN startWifi] SSID ArkiA unable to connect 
[18:12:18.447 startWifi] Wifi stats for ArkiA - signal strength: -54 dBm; Encryption: WPA2_PSK; channel: 1

s60sc, 2024-03-07:
passwords not stored in configs file for security reasons
I cannot help with wifi issues

mamazad7, 2024-03-11:
made it to work, finally
hope my findings may help someone ...
i didn't change the code other than to add my new S3(New)  board got it from Aliexpress as attached. 
those changes were;
-----
in camerapins.h
// added f0r NEW ESP32 S3 from AliExpress 01MAR2024
#elif defined(CAMERA_MODEL_NEW_ESPS3_RE1_0)
#define CAM_BOARD "CAMERA_MODEL_NEW_ESPS3_RE1_0"
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM     10
#define SIOD_GPIO_NUM     21
#define SIOC_GPIO_NUM     14

#define Y9_GPIO_NUM       11
#define Y8_GPIO_NUM       9
#define Y7_GPIO_NUM       8
#define Y6_GPIO_NUM       6
#define Y5_GPIO_NUM       4
#define Y4_GPIO_NUM       2
#define Y3_GPIO_NUM       3
#define Y2_GPIO_NUM       5
#define VSYNC_GPIO_NUM    13
#define HREF_GPIO_NUM     12
#define PCLK_GPIO_NUM     7

#define LED_GPIO_NUM     34
---------
for SD card,  in appGlobals.h

#define SD_MMC_CLK  42
#define SD_MMC_CMD 39
#define SD_MMC_D0 41
--------
MIC pins: have to config in web GUI
WS 37
SCK 36
SD 35

-----
not sure if any changes have to be made for a new board, but works for me 
following is my workaround to make this code to work for me 
for passwords, symbols like # and * don't work (could be more)
but @ and ! works (could be more)
so the best is to start with just 'alphanumeric' before going to a fancy PW with symbols.

a weird experience, most likely just for me....
initially, I tested with my testbench WAP then I changed it to a different WAP. as expected (not really) did not work. after sending a few hours of testing, change the current WiFi CH 11 to CH 1, and start to work again.

[image]

s60sc, 2024-03-11:
Thanks for sharing your findings, I'll add the new board definition in the next release

mamazad7, 2024-03-12:
no worries :) and Thank you.
hope the attached docs help
[SCH_ESP32-S3-Schematic Diagram.pdf](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/files/14566554/SCH_ESP32-S3-Schematic.Diagram.pdf)
[ESP32-S3 MINI_specification.pdf](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/files/14566576/ESP32-S3.MINI_specification.pdf)

s60sc, 2024-03-12:
v9.6.2 released

closed 2024-03-12 by s60sc