s60sc/ESP32-CAM_MJPEG2SD #380: FTP request

sjude68, 2024-03-04:
I want to install the ESP32-Cam in the lift.  I am using the feature to send the recording to FTP server and delete from SD-card. As is no WiFi available on every floor, the recorded file stays on SD-card.  If the Wifi is connected after recording it immediately transfers and deletes the file.

Is there an option to ftp the file as soon as the Wifi is connected when the Auto upload option is enabled.  This will save space on the SD card.

DJprasenjit, 2024-03-07:
I selected auto delete files after FTP. But my 16gb sd card almost full. I deleted all folders from the app. But still 4.8GB used. There should be option to clear all data of sd card

closed 2024-03-12 by s60sc

sjude68, 2024-03-13:
I tried the upload to FTP after the wifi disconnection and reconnection.  It uploads only the last file created after the wifi connection.  The previous motion captured files stay on the SDcard.

Also noticed that version number is not updated on the webpage.  It still shows 9.5.