s60sc/ESP32-CAM_MJPEG2SD #387: ftp uploading after wifi reconnection.

sjude68, 2024-03-13:
I tried the upload to FTP after the wifi disconnection and reconnection. It uploads only the last file created after the wifi connection. The previous motion captured files stay on the SDcard. Can the complete folder be uploaded instead of the latest file only.

Also noticed that version number is not updated on the webpage. It still shows 9.5. Should I also upload the data directory on the SDcard ?

s60sc, 2024-03-13:
All previous files in folder should be uploaded after a new file is created, provided both Auto Upload & Auto SD Delete are selected

>  It still shows 9.5

Then 9.6.2 is not installed

sjude68, 2024-03-13:
I used Ota update to send the compiled bin file.

sjude68, 2024-03-14:
I updated from Arduino directly. It worked. I had to copy the data directory also to the sdcard.

closed 2024-03-14 by sjude68