s60sc/ESP32-CAM_MJPEG2SD #390: [FEATURE REQUEST] Spiff Webserver

rjsachse, 2024-03-18:
would be a good option to be able to select internal flash to serve the website. Can also compress files down using a script like this [one](https://github.com/sidoh/esp8266_milight_hub/blob/master/.build_web.py)

s60sc, 2024-03-18:
As an sd card is required by this app there is no advantage to using internal flash. 
If a user wants to use the app without sd and therefore without recording facility, then in `appGlobals.h`
change:
`#define STORAGE SD_MMC`
to
`#define STORAGE LittleFS`
which will store web files on internal flash, provided the app is compiled with a different partition scheme for ESP32 (not needed for ESP32S3) which Ive added to `extras` folder

rjsachse, 2024-03-19:
I understand that. Just my scenario I have a a couple of cameras at my remote property with no internet. I swap out sd cards when I'm at the property I then have to remember to put the data back on the new card. Would be great if I could just swap out with formated sd cards without worrying about transferring the data to all cards. Would also be easier for my wife to do it too.

rjsachse, 2024-03-29:
Noticed it was requested here before https://github.com/s60sc/ESP32-CAM_MJPEG2SD/issues/126#issue-1362433534

closed 2024-04-06 by s60sc